moononournation/Arduino_GFX #343: CS pin on I2C IO expander for Sensecap Indicator

chalei, 2023-09-06:
Hello @moononournation, I am in the process of testing the library for the Seeedstudio's sensecap indicator.

**it is using ESP32S3 using the st7701 as the screen**
Last night I try the PDQgraphicstest by selecting the ESP32_S3_RGB option  and have a successful result like the video below

https://github.com/moononournation/Arduino_GFX/assets/5271515/b6029b83-6aa6-4468-ab2c-3071c66756f1

this is the setting that I use
```
#define GFX_DEV_DEVICE ESP32_S3_RGB
#define GFX_BL 45
Arduino_DataBus *bus = new Arduino_SWSPI(
    GFX_NOT_DEFINED /* DC */, GFX_NOT_DEFINED /* CS */,
    41 /* SCK */, 48 /* MOSI */, 47 /* MISO */);

// option 1:
// Uncomment for 4" rect display
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    18 /* DE */, 17 /* VSYNC */, 16 /* HSYNC */, 21 /* PCLK */,
    4 /* R0 */, 3 /* R1 */, 2 /* R2 */, 1 /* R3 */, 0 /* R4 */,
    10 /* G0 */, 9 /* G1 */, 8 /* G2 */, 7 /* G3 */, 6 /* G4 */, 5 /* G5 */,
    15 /* B0 */, 14 /* B1 */, 13 /* B2 */, 12 /* B3 */, 11 /* B4 */,
    1 /* hsync_polarity */, 10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
    1 /* vsync_polarity */, 10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, st7701_type1_init_operations, sizeof(st7701_type1_init_operations));
```

and I choose the OPI PSRAM option

but after a while it stop working and only the backlight is turned on until now even after I try to reupload the code.

[image]

Does this have something to do with the CS pin connected to the I2C IO expander?
[image]

I also already try to define the CS pin using the PCA9535 library 

```
#include<Arduino_GFX_Library.h>
#include <PCA95x5.h>

// select one of them
PCA9535 ioex;

#define GFX_BL 45
Arduino_DataBus *bus = new Arduino_SWSPI(
    GFX_NOT_DEFINED /* DC */, PCA95x5::Port::P04 /* CS */,
    41 /* SCK */, 48 /* MOSI */, GFX_NOT_DEFINED /* MISO */);
```

but the result still the same. Is there something wrong with the setting that I use? how to define the CS pin that is connected to the Io expander?

Thank You

moononournation, 2023-09-07:
You may need a custom class, you can refer to this class for reference:

https://github.com/moononournation/Arduino_GFX/blob/master/src/databus/Arduino_XCA9554SWSPI.h

u4mzu4, 2023-09-11:
Please see this disussion about the SenseCap Indicator:
[https://github.com/moononournation/Arduino_GFX/discussions/334](https://github.com/moononournation/Arduino_GFX/discussions/334)

chalei, 2023-09-11:
> Please see this disussion about the SenseCap Indicator: #334

Ahh thank you for this, this is the one I've been lookin for days. Got it working now, and set the rotation to 2 for the right orientation

closed 2023-09-11 by chalei

u4mzu4, 2023-09-11:
@chalei : Use the HW rotation instead of the SW rotation (2), it's only one additional line in the _st7701_type1_init_operations_.