moononournation/Arduino_GFX #357: ESP32 NT35510 PAR 8 Bit crash

KrystofVydra, 2023-10-07:
Crash happens straight after `Arduino_GFX Hello World example` message
Upload settings are: 
[image]

I am using 800 x 480 LCD Display module with NT35510 driver set to 8 bit mode via the module resistors. My connection is as follows:
`
27 /* DC */
5 /* CS */
26 /* WR */
16 /* D0 */
17 /* D1 */
18 /* D2 */
19 /* D3 */
21 /* D4 */
12 /* D5 */
23 /* D6 */
25 /* D7 */
VCC - VIN
GND - GND
`

I am using the HelloWorld.ino example where I added my configuration like this:
`
#include <Arduino_GFX_Library.h>

#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_DataBus *bus = new Arduino_ESP32PAR8(27 /* DC */, 5 /* CS */, 26 /* WR */, GFX_NOT_DEFINED /* RD */,
    16 /* D0 */, 17 /* D1 */, 18 /* D2 */, 19 /* D3 */, 21 /* D4 */, 12 /* D5 */, 23 /* D6 */, 25 /* D7 */);
    
/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
Arduino_GFX *gfx = new Arduino_NT35510(
  bus, 7 /* RST */, 0 /* rotation */);
  
#endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/
`

After flashing the code, the ESP32 keeps rebooting right after printing the `Arduino_GFX Hello World` example message.
The issue persists even if I completely disconnect the display from the board.

The crash log:
`
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13936
load:0x40080400,len:3600
entry 0x400805f0
Arduino_GFX Hello World example
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13936
load:0x40080400,len:3600
entry 0x400805f0
Arduino_GFX Hello World example
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13936
load:0x40080400,len:3600
entry 0x400805f0
Arduino_GFX Hello World example
`

moononournation, 2023-10-07:
seems some loop running too long and triggered the WDT, try comment out the fill command to check is it related.

KrystofVydra, 2023-10-07:
Still getting the same error when commenting out the fillScreen command.

moononournation, 2023-10-07:
what is your backlight connected to? if not connected to gpio, please comment out the GFX_BL define.

Asanatsa, 2023-10-19:
The same thing happened to me. There seems to be some error in init() and it trigger the watchdog

moononournation, 2023-10-23:
What ESP32 module are you using? Is it have PSRAM?

moononournation, 2023-10-24:
GPIO 7 & 8 are used for Flash, so you should avoid use GPIO 7 as RST pin.

closed 2023-10-24 by moononournation

Asanatsa, 2023-10-27:
> What ESP32 module are you using? Is it have PSRAM?

ESP32-WROOM-32 , and there's no PSRAM
I didn't connect anything

moononournation, 2023-10-27:
ok, then please avoid using GPIO 7.

KrystofVydra, 2023-10-27:
The issue is that even If I do not connect anything. Is there any idea? @Asanatsa did you manage to overcome this issue? I was busy last weeks and this weekend I will continue on testing.

KrystofVydra, 2023-10-27:
@moononournation which model of esp32 are you using and what exact software version? is this even possible to work or should I switch to different platform (rp2040 or something else)? Thank you very much

moononournation, 2023-10-27:
ESP32, just avoid using GPIO 7 should be ok.