moononournation/Arduino_GFX #359: Memory Issue with ZX3D95CE01S_TR

rguilbeau, 2023-10-08:
Hi,

First of all, congratulations on your work, your library is very comprehensive!

However, I have a problem: I'm trying to get the ZX3D95CE01S_TR device to work, and in your example, Arduino_GFX is not declared.

https://github.com/moononournation/Arduino_GFX/blob/b0781cb14625be081424e50c80fb2afa0cba2fac/examples/PDQgraphicstest/Arduino_GFX_dev_device.h#L468-L481

I tried declaring it like the ZX3D95CE01S_AR :
```
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, gc9503v_type1_init_operations, sizeof(gc9503v_type1_init_operations));
```

But I'm getting a memory error::

> Arduino_GFX PDQgraphicstest example!
> E (339) lcd_panel.rgb: esp_lcd_new_rgb_panel(151): no mem for frame buffer
> ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4037a938 
> file: ".pio/libdeps/esp32-s3-devkitc-1-n16r8v/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.cpp" line 118
> func: uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)
> expression: esp_lcd_new_rgb_panel(_panel_config, &_panel_handle)
> 
> abort() was called at PC 0x4037a93b on core 1

If I reduce the resolution to 380x380, I no longer have a memory issue, but nothing happens; the screen remains black.

If you have any ideas about the issue, that would be great !

Thanks again,
Regards, Romain.

rguilbeau, 2023-10-08:
Hi again,

I found my first mistake; PSRAM was disabled in the board settings. Therefore, I no longer have the memory issue.

However, I still have the issue of the screen remaining black without any error. I might still have a wrong configuration for my board. Here is my configuration:

[image]

moononournation, 2023-10-09:
ZX3D95CE01S_TR have another declaration, please get update.

rguilbeau, 2023-10-09:
I'm using the latest version (I tried with 1.3.8 and master), if I understand correctly, the declaration in the PDQgraphictest example is not correct, is it?

```
#elif defined(ZX3D95CE01S_TR)
#define GFX_DEV_DEVICE ZX3D95CE01S_TR
#define GFX_BL 5
Arduino_DataBus *bus = new Arduino_SWSPI(
    GFX_NOT_DEFINED /* DC */, 38 /* CS */,
    45 /* SCK or SCLK */, 48 /* MOSI */, GFX_NOT_DEFINED /* MISO */);
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 42 /* HSYNC */, 39 /* PCLK */,
    18 /* R0 */, 8 /* R1 */, 3 /* R2 */, 46 /* R3 */, 10 /* R4 */,
    14 /* G0 */, 13 /* G1 */, 12 /* G2 */, 11 /* G3 */, 16 /* G4 */, 17 /* G5 */,
    45 /* B0 */, 48 /* B1 */, 47 /* B2 */, 0 /* B3 */, 21 /* B4 */,
    1 /* hsync_polarity */, 8 /* hsync_front_porch */, 10 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
    1 /* vsync_polarity */, 8 /* vsync_front_porch */, 10 /* vsync_pulse_width */, 20 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, gc9503v_type1_init_operations, sizeof(gc9503v_type1_init_operations));
```

rguilbeau, 2023-10-09:
I found a related discussion: [#123](https://github.com/moononournation/Arduino_GFX/discussions/329).

Indeed, the declaration of Arduino_RGB_Display is different for the ZX3D95CE01S_TR.

 I won't close the issue because this declaration is missing in PDQgraphicstest/Arduino_GFX_device.h

```
Arduino_RGB_Display *gfx = new Arduino_RGB_Display( 
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush (IPS, in makerfabs project it is IPS) */, 
  bus, 41 /* RST */, gc9503v_type1_init_operations, sizeof(gc9503v_type1_init_operations));
```

Thanks again for your work

moononournation, 2023-10-09:
Thx for check this out, just added back.

rguilbeau, 2023-10-09:
I looked at your commit, the RST pin should be set to 41 (see my previous comment)

moononournation, 2023-10-09:
fixed

rguilbeau, 2023-10-10:
Perfect 👍

closed 2023-10-10 by rguilbeau