moononournation/Arduino_GFX #371: 1.9inch_ESP32-1732S019 HELP

taoloo897, 2023-10-30:
HI. I tried to use your library to upload sketch PDQgraphicstest to this bord 1.9inch_ESP32-1732S019
https://www.aliexpress.com/item/1005005951347460.html?spm=a2g0o.order_list.order_list_main.13.27f118022Qp7UO
but, nothing work, the screen is black and on serial on serial monitor nothing appears 
I must uload original bin with esp tool to make the bord to show  the vendor example .
I tried t_eSPI examples too, i tried T_HMI too , not work.
Please HELP ME TO USE THIS BORD  with lcd with arduino IDE.
THANKS IN ADVANCE

moononournation, 2023-10-30:
In PDQgraphicstest example, simply uncomment #define ESP32_1732S019 will work

taoloo897, 2023-10-30:
Thanks, It works if uncomment that line.
A problem is why not print on serial monitor?
I tried uncomment // Serial.setDebugOutput(true); in PDQgraphicstest.ino but nothing on serial.
Another issue , can you make it works directly with T_eSPI? 
Thanks again for all your time

moononournation, 2023-10-30:
Sorry, this is the page for Arduino_GFX library.

taoloo897, 2023-10-30:
Ok, sorry, my mistake, please tell me why not see on serial monitor what i want to debug, has this board an issue?
But if i upload the factory bin with flash tool it send messages on serial monitor.
Thanks again for your time.

moononournation, 2023-10-30:
related to enable USB CDC option?

taoloo897, 2023-10-31:
AWSOME.  I disabled  USB CDC You are greate . THANKS A LOT,

closed 2023-10-31 by taoloo897