moononournation/Arduino_GFX #372: Is there by any chance any support for elecrow ESP32 7" Display ?

mikrocoder, 2023-10-31:
Hi,

would your lib work with this display as well? [elecrow esp32-display-7](https://www.elecrow.com/esp32-display-7-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html)
This seems to have an exotic Driver IC EK9716BD3 and EK73002ACGB? Or are they compatible with another?
The elecrow support can not help.

debrisapron, 2023-11-06:
Hey @mikrocoder, I'm not the dev but it actually does work, for some reason you have to use v1.2.8. Here's the code I'm using which I dug out of the Elecrow examples:

```
#include <Arduino_GFX_Library.h>
const int BACKLIGHT_PIN = 2;

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */,
    GFX_NOT_DEFINED /* SDA */, 41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */,
    0 /* PCLK */, 14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */,
    45 /* R4 */, 9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */,
    16 /* G4 */, 1 /* G5 */, 15 /* B0 */, 7 /* B1 */, 6 /* B2 */,
    5 /* B3 */, 4 /* B4 */);

Arduino_RPi_DPI_RGBPanel *lcd = new Arduino_RPi_DPI_RGBPanel(
    bus, 800 /* width */, 1 /* hsync_polarity */,
    40 /* hsync_front_porch */, 48 /* hsync_pulse_width */,
    40 /* hsync_back_porch */, 480 /* height */, 1 /* vsync_polarity */,
    13 /* vsync_front_porch */, 1 /* vsync_pulse_width */,
    31 /* vsync_back_porch */, 1 /* pclk_active_neg */,
    16000000 /* prefer_speed */, true /* auto_flush */);

void setup() {
    pinMode(BACKLIGHT_PIN, OUTPUT);
    digitalWrite(BACKLIGHT_PIN, HIGH);
    lcd->begin();
    lcd->fillScreen(RED);
}
```

I'm still trying to find a solution which works with the latest version!

debrisapron, 2023-11-06:
OK I posted too quickly, the solution for 1.3.8 is here: https://github.com/moononournation/Arduino_GFX/issues/373

mikrocoder, 2023-11-09:
Best Thanks.

moononournation, 2023-11-16:
If you found the correct parameters for this dev device, please provide the PDQgraphicstest Youtube link. I will add the parameters to the example and share the link in README.md.

mikrocoder, 2023-11-23:
Hello,

I only got the original demo to compile until today and I can flash it. Everything else either does not compile or the display remains black. The example from debrisapron compiles but the display stays black. The wiki from elecrow and their examples is the last thing I have seen so far. None of it works. It also seems to only compile up to ESP core version 2.0.3. If you use a newer core version, nothing compiles. The libraries are also unclear. There is an SPI and SD lib from elecrow. What should you do with them? The Arduino IDE already includes them. Should you overwrite them and thus become incompatible with the rest of the Arduino world? I will try out the example from Thro42 ... Thanks for that.

Thro42, 2023-11-24:
@mikrocoder I known your pain. I Update all and the the Example PDQgraphicstest works with.
` #define ESP32_8048S070`
And this settings:
[image]
Sorry for les doku. of my Project. It is fresh an I work on it.

closed 2023-11-30 by moononournation