moononournation/Arduino_GFX #373: Display declaration for library version >1.3.1?

DrTron, 2023-10-31:
Hello!
I recently acquired an ESP32-8048S050 (800x480 IPS) display. The vendor provided example sketches, whose display declaration works up to library version 1.3.1. Above that, they don't, and it seems that the display declaration has changed, although the changelogs don't mention that.

I'm now trying to create a valid declaration for the current library version, but somehow can't figure out what I may be missing.
The "old" declaration is this:
```Arduino_GFX *gfx = create_default_Arduino_GFX();
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */
);
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
   bus,
   800 /* width */, 0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
   480 /* height */, 0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
   1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);
```

whereas the "new" declaration, as taken from the ESP32-8048S070 example, needs to be something like this:
```
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);
```

But even with the values taken from the "old" sample, the sketch compiles but the display stays dark.
Any ideas?

Thanks!

moononournation, 2023-10-31:
You should fill your gpio number for each pin. E.g. your DE pin should be 40 instead of 41.

DrTron, 2023-10-31:
Yes, I thought I did exactly that yesterday. Tried again today with 1.3.8 and now it works.
Must have had a typo somewhere.

Thanks, and sorry for the bother!

closed 2023-11-16 by moononournation