moononournation/Arduino_GFX #381: any possible fixes for display shifting during OTA on ESP32-S3 RGB panel display?

alpha815, 2023-11-15:
### Issue
Display Shift During OTA on ESP32-S3 RGB Panel (ZX3D95CE01S-TR-4848)

### Description
When performing Over-The-Air (OTA) updates on ESP32-S3 with OPI SRAM using the Arduino IDE (ESP32 board version 2.0.14, Library version 1.3.8), the display on the ZX3D95CE01S-TR-4848 starts shifting to the right. The issue seems to occur during the OTA process and is suspected to be related to the SPI bus, which is shared with SPI RAM.

### Environment
- Microcontroller: ESP32-S3 with OPI SRAM
- IDE: Arduino IDE
- ESP32 Board Version: 2.0.14
- Library Version: 1.3.8
- Display Board: ZX3D95CE01S-TR-4848
- Display Sample code: [GitHub Link](https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848)

### Issue Details
- **Issue Link:** [YouTube Video](https://youtube.com/shorts/TwakucdvEQY?feature=share)
- **Initialization Code:** [Display Initialization Code](https://github.com/alpha815/ZX3D95CE01S-TR-4848_SAMPLE/blob/main/LVGL_ZX3D95CE01S-TR-4848_SHT20/display_config.h)

moononournation, 2023-11-16:
You may try to find the correct hsync and vsync value in display datasheet. It is variant for each RGB display.

closed 2023-11-16 by moononournation

alpha815, 2023-11-16:
> You may try to find the correct hsync and vsync value in display datasheet. It is variant for each RGB display.

Hi, thanks you, but i don't think it is related to display configuration. display works fine without any problem. this shifting issue only appear during OTA.