moononournation/Arduino_GFX #387: 使用st7789（240*320）展示20*20的jpg图片出现黑色区域

qiaobaaa, 2023-11-29:
hi您好，在使用例程Imgviewerjpeg时候，从SD卡读取一张20*20像素的jpeg图片并显示，但是这块串口打印出来了两次，两次pDraw->iWidth, pDraw->iHeight的结果分别是size=32*16，size=32*4，导致图片虽然完整显示了，但是多出了了一块12*20的黑色区域，实际图片像素区域只有20*20，这个问题不知道该怎么解决，可以只显示20*20的像素区域吗

static int jpegDrawCallback(JPEGDRAW *pDraw)
{
   Serial.printf("Draw pos = %d,%d. size = %d x %d\n", pDraw->x, pDraw->y, pDraw->iWidth, pDraw->iHeight);
  gfx->draw16bitBeRGBBitmap(pDraw->x, pDraw->y, pDraw->pPixels, pDraw->iWidth, pDraw->iHeight);
  return 1;
}

moononournation, 2023-11-29:
This example is using JPEGDEC decode library. I think you should report it to JPEGDEC author.

qiaobaaa, 2023-11-29:
> This example is using JPEGDEC decode library. I think you should report it to JPEGDEC author.

非常感谢，这似乎是jpeg在解析过程中的一种保护策略，像素数组申请的大小要比实际多一点，全屏的时候并不影响，只显示一部分的图片显示就会存在问题

qiaobaaa, 2023-11-29:
> This example is using JPEGDEC decode library. I think you should report it to JPEGDEC author.

您好问题解决了感觉很神奇，发现使用photoshop生成的照片20 * 20像素就不会出现这个问题（图片保存质量高及其以上），之前处理图片都是用win10自带的画图工具去处理，从上到下依次是ps处理的：最差质量图片，中质量图片，高质量图片，最佳质量图片，画图工具处理的图片，很明显在一些图片就是会出现黑色区域，建议您可以在代码注释里加上这块，后面人就能避免这个问题了
[image]

[image]

closed 2023-11-30 by moononournation