moononournation/Arduino_GFX #391: ESP-IDF gives the error "GPIO was not declared in this scope" when compiling with ESP-IDF v5.1

ergodone, 2023-12-01:
When compiling GFX_Library_for_Arduino ESP-IDF gives multiple errors "GPIO was not declared in this scope" when compiling with ESP-IDF v5.1

E.G.
libraries/GFX_Library_for_Arduino/src/databus/Arduino_ESP32SPI.cpp:183:30: error: 'GPIO' was not declared in this scope

See: https://github.com/espressif/esp-idf/issues/9184

moononournation, 2023-12-01:
Expected, Arduino_GFX only tested in Arduino IDE.

ergodone, 2023-12-01:
It is with the Arduino IDE with the ESP32 package v3.0.0-alpha2

moononournation, 2023-12-01:
not support alpha version

ergodone, 2023-12-01:
That's fine.  It was more to make people aware of the upcoming issue.  I didn't see any mention of it.

moononournation, 2023-12-02:
I will make relavant change after it release.

closed 2023-12-02 by moononournation

lamfe, 2025-05-13:
Latest ESP32 package release is now 3.2. This issue is relevant now.