moononournation/Arduino_GFX #392: GFXfont support broken?

aremmell, 2023-12-02:
This has a question mark after it, because I am not 100% sure that I or Arduino IDE are not somehow to blame, but:

I am setting a `GFXfont*` (which I verify the validity of right before I pass it in to `Arduino_Canvas::setFont`), then calling `getTextBounds`.

I kept getting 0 back for width and height, so I stuck a bunch of serial logging in there and ran it again.

Turns out that all of the properties of the `GFXfont` are as they were in the constructor except for `GFXfont::last`; that is set to 0, every time. This causes the `charBounds` function to essentially be a NOOP since `ch >= first && ch <= last` (Arduino_GFX.cpp:2633) is now false.

I have tried several fonts from Adafruit's GFX library. I have uninstalled and reinstalled both your library and Adafruit's. I cannot locate any code in your library that writes to that property.

I have tried with another board to make sure it wasn't the hardware (although it could still be the board type; they are both ESP32-S3 based).

At Arduino_GFX.cpp:2632, last == 0, but all the other properties of the font are intact. I have an ESP32-S3 dev kit here, so I will probably get that out and see if I can glean anything from the debugger. In the mean time I thought you should be aware of this potential issue.

Arduino GFX: 1.4.0
Arduino IDE: 2.2.1 (macOS)
Boards: Adafruit Qualia RGB 666 ESP32-S3, UM ProS3

aremmell, 2023-12-02:
Cancel. I just compiled and ran with PlatformIO instead, and it works fine. Don't ask me to explain how this kind of thing is possible, because I have no idea. The library versions are identical, so that leaves the compiler and linker.

closed 2023-12-02 by aremmell

moononournation, 2023-12-02:
I still stick at Arduino IDE 1.8.9 on development. Since I heard 2.x not support Java.