moononournation/Arduino_GFX #399: Using the Wire bus - some topics to be discussed

mathertel, 2023-12-11:
Hi ,
I just make the TWI/Wire bus available to this library as an interface for displays and suggest some additions to the base databus class.
Before merging from <https://github.com/mathertel/Arduino_GFX> I like your feedback...

To switch between commands and data no extra GPIO or 9-th bit is used. Instead the chip driver implementation knows what prefix byte is used for the transfer. This must be given to the bus implementation.

I added a function `virtual void setOption(option_key option_key, uint16_t option_value){};` to the databus class. The Wire class overrides this function to record the wire specific parameters. The other busses can ignore this and the base class implements an empty stub for it:

```cpp
  _bus->setOption(Arduino_DataBus::TWI_COMMAND_PREFIX, SSD1306_COMMAND);
  _bus->setOption(Arduino_DataBus::TWI_DATA_PREFIX, SSD1306_DATA);
```

I don't want to add this to the bus creation as a parameter as it is protocol knowledge from the driver chip datasheet and must not be known on the interfaces.

I found no way for the driver to find out what bus interface is used and only transmit the prefixes in case of a Wire bus.
So some additional work must be done to make the driver class independant from the bus specifics and move this into the wire implementation.

The TWI_BUFFER_LENGTH from Arduino_databus.h seems not to be used anywhere but also the size of the Wire buffer is not equal 32 in the ESP SoCs.

```CPP
#if defined(I2C_BUFFER_LENGTH)
#define TWI_BUFFER_LENGTH I2C_BUFFER_LENGTH
#else
#define TWI_BUFFER_LENGTH 32
#endif
#endif
```

WIRE transfers have to respect this size and when transferring data a new packet must be initialized in writePixels.
I added a writePixels(uint8_t *data, ...) in addition to the 16-bit version.
Maybe it should be named `writePixels8` or `writePixelsBytes` ?

I have not finished working on the 1-bit OLED displays as I like to make it work with the SPI bus as well.
But the Wire approach is working for now.

Please give me feedback. Thanks

moononournation, 2023-12-12:
someone already merged a Arduino_Wire class, do you want to do similar thing?

mathertel, 2023-12-12:
That was me. But the implementation has some limits as mentioned above.
I ask as I do changes to the common coding in the base databus class. 

Do you mind when I just continue ?

moononournation, 2023-12-12:
The base class is shared by many class and works well now. I don't think it is a good idea to change it.

mathertel, 2023-12-12:
all I add are 2 empty virtual functions for the reasons mentioned above:

```cpp
virtual void setOption(option_key option_key, uint16_t option_value){};
virtual void writePixels(uint8_t *data, uint32_t len){};
```
that should not harm existing bus implementations.

closed 2024-01-05 by moononournation