moononournation/Arduino_GFX #402: Non default fonts not showing when canvas used within a class

rdekens, 2023-12-14:
### Discussed in https://github.com/moononournation/Arduino_GFX/discussions/400

when using a master canvas - and sub canvasses - 
ie master canvas as screen buffer , and then other canvases set with  (GFX_SKIP_OUTPUT_BEGIN), 

the sub canvasses do not show anyhting other than the **default font**
ie text printed after setFont() is shown,
text printed with setFont(<font of choice>)  doesnt show - 
drawing commands work ok  - at least recatnagels and lines - havent check circles and triangles at the moement - but there is at least inconsistent behaviour with the fonts ...

[HelloWorldGfxfont_flat.zip](https://github.com/moononournation/Arduino_GFX/files/13679169/HelloWorldGfxfont_flat.zip)

rdekens, 2023-12-16:
in  Arduino_GFX.cpp
size_t Arduino_GFX::write(uint8_t c)
{
#if !defined(ATTINY_CORE)
  if (gfxFont) // custom font
  {
    if (c == '\n') // Newline
    {
      cursor_x = _min_text_x; // Reset x to zero, advance y by one line
      cursor_y += (int16_t)textsize_y * (uint8_t)pgm_read_byte(&gfxFont->yAdvance);
    }
    else if (c != '\r') // Not a carriage return; is normal char
    {
      uint8_t first = pgm_read_byte(&gfxFont->first),
              last = pgm_read_byte(&gfxFont->last);
      **if (1) // ##### ((c >= first) && (c <= last)) // Char present in this font?**
      {
        GFXglyph *glyph = pgm_read_glyph_ptr(gfxFont, c - first);
        uint8_t gw = pgm_read_byte(&glyph->width),
                xa = pgm_read_byte(&glyph->xAdvance);
        int16_t xo = pgm_read_byte(&glyph->xOffset);
        if (wrap && ((cursor_x + ((xo + gw) * textsize_x) - 1) > _max_text_x))
        {
          cursor_x = _min_text_x; // Reset x to zero, advance y by one line
          cursor_y += (int16_t)textsize_y * (uint8_t)pgm_read_byte(&gfxFont->yAdvance);
        }
        drawChar(cursor_x, cursor_y, c, textcolor, textbgcolor);
        cursor_x += (int16_t)textsize_x * xa;
      }

    }

appears to be related to reding the first and last of the font for some reason - if I force it to be "inside " the font then it works - 
..

rdekens, 2023-12-16:
ok - 
somewhere along the line - 
Adafruit library changed to ? 
/// Data stored for FONT AS A WHOLE
typedef struct
{
	uint8_t *bitmap;	///< Glyph bitmaps, concatenated
	GFXglyph *glyph;	///< Glyph array
	uint16_t first;		///< ASCII extents (first char)
	uint16_t last;	   ///< ASCII extents (last char)
	uint8_t yAdvance; ///< Newline distance (y axis)
} GFXfont;

your library still has >
typedef struct
{
	uint8_t *bitmap;	///< Glyph bitmaps, concatenated
	GFXglyph *glyph;	///< Glyph array
	uint8_t first;		///< ASCII extents (first char)
	uint8_t last;	   ///< ASCII extents (last char)
	uint8_t yAdvance; ///< Newline distance (y axis)
} GFXfont;

I imaging the difference in in supporting unicode -  - orat least the intent to - 
the reads of the side end up incorrect and hence the fort doesn't get printed.

looks like a simple headerr change should fix it.
Ray

moononournation, 2024-01-24:
Thx for let me know it. Just revised GFXfont struct, please reopen if still have problems.

closed 2024-01-24 by moononournation

rdekens, 2024-01-29:
didn't work on mine -
attached is the differences between mine and 1.4.2 - 
[GFX_Diffs.zip](https://github.com/moononournation/Arduino_GFX/files/14082408/GFX_Diffs.zip)