moononournation/Arduino_GFX #408: Databus lines on Expander (Xl9535 or similar)

s54b32julian, 2023-12-27:
I must move from the databus call "Arduino_ESP32RGBPanel" to a custom databus call, where i put the 18 (16) datalines on a port expander (XL9535), due the lack of GPIOs in my project PCB. Can you guide me in de correct direction, i´m a bit lost on this.

Any help is appreciated!

moononournation, 2023-12-27:
no, rgb display cannot use expander

s54b32julian, 2023-12-28:
Why? Is this a technical limitation and therefore simply impossible or can it be implemented? If possible and you have interest in this, i could sponsor you for this feature. Please let me know!

moononournation, 2023-12-28:
The RGB display requires very fast continuous data stream and it is tightly coupled with ESP32-S3. I just calling the ESP32-S3 underlying API.

closed 2024-01-05 by moononournation