moononournation/Arduino_GFX #410: Arduino ESP32 alpha 3.0.x, GFX_Library_for_Arduino\src/Arduino_DataBus.h:182:13: error: 'i80_device_list' has not been declared

6bigfire, 2023-12-30:
When I compile with Arduino esp32 alpha 3.0.x, I have got the errors. Can anyone help me? Thank you!

In file included from d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:4,
                 from D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\lv_screen.hpp:34,
                 from D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers.h:7,
                 from D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\openai.ino:1:
d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:182:13: error: 'i80_device_list' has not been declared
  182 |   LIST_HEAD(i80_device_list, lcd_panel_io_i80_t)
      |             ^~~~~~~~~~~~~~~
d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:182:3: error: ISO C++ forbids declaration of 'LIST_HEAD' with no type [-fpermissive]
  182 |   LIST_HEAD(i80_device_list, lcd_panel_io_i80_t)
      |   ^~~~~~~~~
d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:182:48: error: expected ';' at end of member declaration
  182 |   LIST_HEAD(i80_device_list, lcd_panel_io_i80_t)
      |                                                ^
      |                                                 ;
d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:183:3: error: 'device_list' does not name a type
  183 |   device_list; // Head of i80 device list
      |   ^~~~~~~~~~~
d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:217:3: error: ISO C++ forbids declaration of 'LIST_ENTRY' with no type [-fpermissive]
  217 |   LIST_ENTRY(lcd_panel_io_i80_t)
      |   ^~~~~~~~~~
d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:217:32: error: expected ';' at end of member declaration
  217 |   LIST_ENTRY(lcd_panel_io_i80_t)
      |                                ^
      |                                 ;
d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:218:3: error: 'device_list_entry' does not name a type
  218 |   device_list_entry; // Entry of i80 device list
      |   ^~~~~~~~~~~~~~~~~
In file included from d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:19:
d:\source\Projects_SquareLine\export_Arduino\RFID_PAD\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:43:3: error: 'esp_lcd_rgb_panel_frame_trans_done_cb_t' does not name a type; did you mean 'esp_lcd_panel_io_color_trans_done_cb_t'?
   43 |   esp_lcd_rgb_panel_frame_trans_done_cb_t on_frame_trans_done; // Callback, invoked after frame trans done
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   esp_lcd_panel_io_color_trans_done_cb_t

moononournation, 2023-12-30:
master branch fixed this

6bigfire, 2024-01-01:
thanks lot!

6bigfire, 2024-01-03:
Board: ESP32S3 DEV MODULE
IDE: Arduino IDE 2.2.1
Board Lib version: 3.0.0-alpha3
GFX version: master

I have got new mistake. like this:

In file included from D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers.h:7,
                 from D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\openai.ino:1:
D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\lv_screen.hpp:38:1: error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32PAR8Q'?
   38 | Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
      | ^~~~~~~~~~~~~~~~~~~~~
      | Arduino_ESP32PAR8Q
D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\lv_screen.hpp:46:1: error: 'Arduino_RGB_Display' does not name a type
   46 | Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
      | ^~~~~~~~~~~~~~~~~~~
In file included from D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\lv_screen.hpp:55:
D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\touch.h: In function 'bool touch_touched()':
D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\touch.h:166:71: error: 'gfx' was not declared in this scope
  166 |     touch_last_x = map(ts.points[0].x, TOUCH_MAP_X1, TOUCH_MAP_X2, 0, gfx->width() - 1);
      |                                                                       ^~~
D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\lv_screen.hpp: In function 'void my_disp_flush(lv_disp_drv_t*, const lv_area_t*, lv_color_t*)':
D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\lv_screen.hpp:73:3: error: 'gfx' was not declared in this scope
   73 |   gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
      |   ^~~
D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\lv_screen.hpp: In function 'bool lv_screen_init()':
D:\source\Projects_SquareLine\export_Arduino\RFID_PAD\module_test\openai\src\drivers\lv_screen.hpp:113:3: error: 'gfx' was not declared in this scope
  113 |   gfx->begin();
      |   ^~~

closed 2024-01-05 by moononournation

jesetek, 2024-05-22:
Do we know when this will be fixed?
I have the same issue on 3.0.0-rc3 :(

buccaneer-jak, 2024-06-07:
> yes, alpha 3.0.x not yet support RGBPanel

Would be really useful fix for the ESP32-C6 boards.