moononournation/Arduino_GFX #411: Arduino AVRPAR16 is missing LITTLE_FOOT_PRINT Virtual Function Overrides when using ARDUINO_AVR_MEGA2560

dapostol73, 2024-01-05:
Got the following Error when using an ARDUINO_AVR_MEGA2560 and Arduino_AVRPAR16.  Looking into it seems the LITTLE_FOOT_PRINT virtual functions are not defined for the AVRPAR & SWPAR classes.   I manage to get around it by adding a #define LITTLE_FOOT_PRINT to my code.  Not sure if this is indented behavior or a missed edge case.

`error: invalid new-expression of abstract class type 'Arduino_AVRPAR16'
   Arduino_DataBus *bus = new Arduino_AVRPAR16(38 /* DC */, 40 /* CS */, 39 /* WR */, 43 /* RD */, 3 /* PORT LOW */, 1 /* PORT HIGH */);`

moononournation, 2024-01-06:
latest should have fixed this.

closed 2024-01-06 by moononournation

dapostol73, 2024-01-08:
When you say latest, what do you mean, latest source or release?  I tested with latest release (1.4.1) and the issue still occurred.