moononournation/Arduino_GFX #413: Graphical Issues when running NT35510 Shield on Mega2560

dapostol73, 2024-01-08:
Was testing the released version of the library at 1.4.1 with the NT35510 Shield and it is having graphical issues.  It looks like the Text writing is failing all together and the other graphical test have alignment issues.  I upload 2 videos to YouTube to see the results.  I use the follow code to enable the bus and gfx to run the PDQgraphicstest

```
#define LITTLE_FOOT_PRINT 
#include <Arduino_GFX_Library.h>

Arduino_DataBus *bus = new Arduino_AVRPAR16(38 /* DC */, 40 /* CS */, 39 /* WR */, 43 /* RD */, 3 /* PORT LOW */, 1 /* PORT HIGH */);
Arduino_GFX *gfx = new Arduino_NT35510(bus, 41, 1);
```
https://youtu.be/c5WUq3HXsCg
https://youtu.be/U9Z0nqgzGXw

moononournation, 2024-01-08:
You mean previous version did not have this issue?

dapostol73, 2024-01-08:
Did some testing with the Arduino IDE looks like 1.3.8 was the last one I could get to work, couldn't test 1.3.9 as it wouldn't compile and 1.4.0 was also broken for me.

dapostol73, 2024-01-09:
I was able to comment our library that was failing on release 1.3.9 and can confirm the issue started with release version 1.3.9
Version 1.3.8 release works fine.

moononournation, 2024-01-10:
Ok, may be located the problems. Would you please help me test with the master branch?

moononournation, 2024-01-10:
while you confirm ok

dapostol73, 2024-01-10:
Confirmed, OK!

moononournation, 2024-01-15:
1.4.2 released.

closed 2024-01-15 by moononournation