moononournation/Arduino_GFX #415: latest version restarts

javidombla, 2024-01-13:
The latest version restarts my ESP32-8048S043C board

moononournation, 2024-01-15:
more details? what's mean latest version? core or library?

javidombla, 2024-01-17:
When updating the library from version 1.4.1 to 1.4.2
When trying to load png images it restarts.
Thanks for your help

moononournation, 2024-01-24:
[image]
Just tested with PNG example without problems.

closed 2024-01-24 by moononournation

javidombla, 2024-02-04:
> 4DF54847-5565-48CD-9E75-A8434F3EF295.jpeg (view on web)
> Just tested with PNG example without problems.

Sorry for the inconvenience. The problem I have is that with the new
version 1.4.1 it is not taking the measurements or positions correctly
before this code:

/* More data bus class:
https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
//Arduino_DataBus *bus = create_default_Arduino_DataBus();

/* More display class:
https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
//Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 0 /* rotation */,
false /* IPS */);

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
  45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
  5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
  8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
  0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width
*/, 8 /* hsync_back_porch */,
  0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width
*/, 8 /* vsync_back_porch */,
  1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true
);

// ST7262 IPS LCD 800x480
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
  800 /* width */, 480 /* height */, rgbpanel);

-------------

  gfx->fillScreen(BLACK);
  gfx->setFont(&number_font20pt7b);
  gfx->setTextSize(1);
  gfx->setTextColor(WHITE);
  gfx->setCursor(5, 50);
  gfx->print("Conectando a la red Wifi");
  gfx->setCursor(5, 100);
  gfx->print(preferences.getString("API_Server", ""));
  gfx->setCursor(5, 150);
  gfx->print(WiFi.localIP());
  gfx->setCursor(5, 200);
  gfx->print(preferences.getString("EQUIPO", ""));
  delay(1500);

The text appeared stuck to the left of the screen and now it appeared
almost in the middle. I attached a photo. When three PNG images are loaded
and the positions do not match, the last one goes off the screen.

Again, I thank you for your attention.