moononournation/Arduino_GFX #417: ESP32-8048S070/ESP32-8048S050 origin dislocation

smily77, 2024-01-15:
I have a ESP32-8048S070 which shows a strange origin dislocation. With draw and and print (TextSize(1)) the origin appears somewhere in the middle of the screen. Funny is that the origin for the print (but only for the print) works correct with textSize(2).
[image]
I use ArduinoIDE 1.8.19 / Arduino GFX 1.4.2 / ESP 2.0.14 (ESP2.0.3 doesn't compile in my IDE with GFX 1.4.2??)
The program:
``#include <Arduino_GFX_Library.h>
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800/* width */, 480/* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

void setup() {
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);

  gfx->begin();
  gfx->setRotation(0);
  gfx->fillScreen(BLACK);
  gfx->drawLine(0, 0, 400, 400, YELLOW);
  gfx -> setCursor(0,0);
  gfx->setTextSize(1);
  for(int i=0;i<26;i++) {
    gfx-> println(i);
  }
  gfx -> setCursor(0,0);
  gfx->setTextSize(2);
  for(int i=0;i<26;i++) {
    gfx-> print("  ");
    gfx-> println(i);
  }
}

void loop() {
  delay(100);
}``
Both, the ESP32-8048S070 & ESP32-8048S050 (5 inch version) show the same error but the displaced isn't exactly at the same place  
BTW: For the ESP32-8048S070&50 the line     1 /* pclk_active_neg */, 16000000 /* prefer_speed */); is required but not mentioned in the Wiki (please update)

DrTron, 2024-01-15:
Same issue here with v. 1.4.2 of the library, using a ESP32-8048S050 display with an ESP32S3 module. Driver is a ST7262.

The origin is somewhere in the upper left quadrant.

For me, though, it doesn't have any impact if that line is there or not.

Alunbb, 2024-01-19:
I can repeat this with: 

1. Elecrow 4.3" RGB 480x272 display
2. Espressif-S3-LCD-EV-Board

I'm trying to run the PDQgraphicstest code using:

- Arduino 2.2.1
- GFX 1.4.2

The graphics origin appears mid way down the screen in the middle of the line.
The code bombs, presumably because memory is getting overwritten when it shouldn't.
Text appears to be fine.

Thanks

Alun

closed 2024-01-24 by moononournation

DrTron, 2024-01-25:
No, lower speed does not change anything.

Alunbb, 2024-01-26:
I have dropped back to using GFX 1.40.

This works for me on the Elecrow 4.3" RGB 480x272 display.
ESP32-S3

moononournation, 2024-02-05:
Please help to test with v1.4.3

smily77, 2024-02-05:
1.4.3 works in my case