moononournation/Arduino_GFX #418: LilyGo T-RGB (2.8) 'error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?'

IAmOrion, 2024-01-15:
I recently got a LilyGo T-RGB the 2.8" version.

I've used what was specified in the docs: 

`#define GFX_EXTRA_PRE_INIT()                                    \
  {                                                             \
    Wire.begin(8 /* SDA */, 48 /* SCL */, 800000L /* speed */); \
  }
#define GFX_BL 46
Arduino_DataBus *bus = new Arduino_XL9535SWSPI(8 /* SDA */, 48 /* SCL */, 2 /* XL PWD */, 3 /* XL CS */, 5 /* XL SCK */, 4 /* XL MOSI */);
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    45 /* DE */, 41 /* VSYNC */, 47 /* HSYNC */, 42 /* PCLK */,
    21 /* R0 */, 18 /* R1 */, 17 /* R2 */, 16 /* R3 */, 15 /* R4 */,
    14 /* G0 */, 13 /* G1 */, 12 /* G2 */, 11 /* G3 */, 10 /* G4 */, 9 /* G5 */,
    7 /* B0 */, 6 /* B1 */, 5 /* B2 */, 3 /* B3 */, 2 /* B4 */,
    1 /* hsync_polarity */, 50 /* hsync_front_porch */, 1 /* hsync_pulse_width */, 30 /* hsync_back_porch */,
    1 /* vsync_polarity */, 20 /* vsync_front_porch */, 1 /* vsync_pulse_width */, 30 /* vsync_back_porch */,
    1 /* pclk_active_neg */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, st7701_type4_init_operations, sizeof(st7701_type4_init_operations));`

and instantly getting errors:

`Clock:54:1: error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?
   54 | Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
      | ^~~~~~~~~~~~~~~~~~~~~
      | Arduino_ESP32S2PAR8Q
Clock:62:1: error: 'Arduino_RGB_Display' does not name a type
   62 | Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
      | ^~~~~~~~~~~~~~~~~~~`
      
      What am I doing wrong? 

MTIA

moononournation, 2024-01-15:
use stable version arduino-esp32 core

IAmOrion, 2024-01-15:
> use stable version arduino-esp32 core

this one?

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

moononournation, 2024-01-15:
yes

closed 2024-01-24 by moononournation