moononournation/Arduino_GFX #420: Using LVGL causes display to mirror over the y-axis

bslisowski, 2024-01-18:
I am using an ESP32-S3 with an ST7262 display. Using the library without LVGL works as intended. 

[image]

I found that changing the direction this loop iterates fixes the mirroring issue, but it causes these vertical lines to appear.

https://github.com/moononournation/Arduino_GFX/blob/5c4e7e626379c6a52ef4dbe5d960f551312f8b35/src/Arduino_G.cpp#L70-L73

[image]

moononournation, 2024-01-24:
Can you should your declaration code?

bslisowski, 2024-01-24:
Ok, so I figured it out and it is not a problem with your library. We have two of these displays and for some reason the second is mirrored, so it is definitely an issue with the display.

closed 2024-01-24 by bslisowski