moononournation/Arduino_GFX #421: How can I write a command with 3 or more bytes using the DataBus API?

mathertel, 2024-01-19:
Using the `writeCommand()` function I can write a single Byte command.
Using the `writeCommand16()` function I can write a 2 byte command.

The sequence 

```cpp
_bus->writeCommand(cmd); // command
_bus->writeCommand(p1);  // first parameter
_bus->writeCommand(p2); // second parameter
```

toggles the DC line between the bytes and corrupts the protocol.

moononournation, 2024-01-24:
Currently no display requires this, so not supported yet. What is your use case?

mathertel, 2024-01-24:
The SSD1306 has this use-case when using SPI for communication to set the write pointer address.
The 3 bytes for these commands must be written in one transmission with CS low for all 3 bytes.
writeC8D16 toggles the line after the first byte and therefore cannot be used.

Here is the code with manually hard coded CS line control:

```cpp
digitalWrite(D1, LOW);
_bus->write(SSD1306_PAGEADDR);
_bus->write(0x00);
_bus->write(0xFF);

// digitalWrite(D1, HIGH);
// digitalWrite(D1, LOW);

_bus->write(SSD1306_COLUMNADDR);
_bus->write(_colStart);
_bus->write(_colEnd);
digitalWrite(D1, HIGH);
```

In theory you can set cs high between the 2 commands but that is not required by the chip.
As long as the CS line is low you can write commands and it must stay low until the command and parameters are sent.

* maybe something like `virtual void writeCommands(uint8_t *commands, uint32_t len) {};` ? 
* On SPI this will pull the CS to low and transfer all given command bytes and release CS.
* On I2C this will start a command sequence and then transfer all given command bytes.
* batchOperation case WRITE_COMMANDS: ... similar to WRITE_BYTES

moononournation, 2024-01-26:
is it can solve this by adding a writeCommandBytes() function to DataBus class?

mathertel, 2024-01-26:
yes right. This is welcome:

```cpp
virtual void writeCommandBytes (uint8_t *data, uint32_t len) { ... };
```

I can provide a first pull request for this change.

mathertel, 2024-01-30:
See #428

closed 2024-01-30 by mathertel