moononournation/Arduino_GFX #430: setcursor X-axis spaces

AndreyUshak, 2024-02-02:
display 5" 800x480
Arduino_ESP32RGBPanel
esp32 2.0.7
ide 1.8.19

Hello
when using gfx fonts
the entire tech is shifted along the X axis by one value,
I ran the HelloWorldGfxfont example and
changed the line
gfx->setCursor(random(gfx->width()), random(gfx->height()));
on gfx->setCursor(0, random(480));
The U8g2Font example shows correctly.
Tell me what the problem is, thank you.
[image]

moononournation, 2024-02-05:
I just released 1.4.3 and fixed some bugs. Please help to check is it also fix your problems.

AndreyUshak, 2024-02-05:
Hi, now everything is fine, everything is in its place,
I want to transfer my project from 480x320(tft_espi) to 800x480,
thank you, very versatile library, good luck.
[image]

closed 2024-02-05 by AndreyUshak