moononournation/Arduino_GFX #431: ImgViewerMjpeg draws with heavy red hue on Giga R1 and GigaDisplay

GreenEnergy1958, 2024-02-03:
I can get Mjpeg files to draw on the GigaDisplay, but with red hue. Added an SD card module, because the Giga's onboard USB using Mbed5 File structure was giving much trouble. Same problem with ImgViewerJpeg.ino. 

These are the libraries in my code:

#include <Arduino_GigaDisplay_GFX.h>
#include "ArduinoGraphics.h"
//#include <Arduino_USBHostMbed5.h>
//#include <DigitalOut.h>
//#include <FATFileSystem.h>
//#include "SD.h"
#include <SdFat.h>
#include "JPEGDEC.h"
#include "MjpegClass.h"

moononournation, 2024-02-03:
I think you are not using my library, please direct raise issue to library author for better support.

GreenEnergy1958, 2024-02-03:
Thank you, another person with same problem opened the issue on JPEGDEC git.

moononournation, 2024-02-03:
I mean Arduino_GigaDisplay_GFX is not belongs to this library, please report your issue to correct author.

closed 2024-02-03 by moononournation