moononournation/Arduino_GFX #432: compatibility with esp32 and weaveshare lcd

rozrabiak, 2024-02-06:
Hi!
Is this library compatible with:
LoLin32 (pinout below)
chip ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz

And this LCD:
https://www.waveshare.com/wiki/4inch_TFT_Touch_Shield

I can't display anything using ESP32 :-(
With Arduino UNO works fine.

[image]

moononournation, 2024-02-06:
The display module designed for 5V MCU may not work for ESP32.

rozrabiak, 2024-02-07:
But from Weaveshare (link in my first post):

"STM32
The demo has been tested on XNUCLEO-F103RB, just insert XNUCLEO-F103RB as shown below. The model of XNUCLEO-F103RB is STM32F103RBT6. If you need to port the program, please connect it according to the actual pin and the schematic diagram."

STM32F103RBT6 accepts voltage: **2 - 3.6V**

moononournation, 2024-02-07:
seems it is not an issue

rozrabiak, 2024-02-07:
OK, so, what I can do?
Wires checked many times, LoLin32 pins checked.
On Arduino UNO above LCD works.
Now Im reading Waveshare product page and there is information: **I/O VOLTAGE: 3.3V/5V**
DIP switches changed on LCD..

I have no idea..

closed 2024-02-22 by moononournation