moononournation/Arduino_GFX #433: `Arduino_ESP32SPI`: `startWrite` sets CPOL/CPHA too late

and3rson, 2024-02-06:
# Preconditions

I'm using ESP32-C3.

I'm trying to share SPI bus between SPI card (mode 0) and ST7789 display (mode 3).

# Problem

The SPI mode is not set properly when I call any drawing functions directly after working with SD card (which changes SPI mode to 0).
Consider this timing diagram:
[image]
As you can see, **SCK polarity is initially 0 (mode 0, set by SD card) and stays low even after display is selected (/LCD is asserted).**

This leads to the first write being misunderstood by the display and simply ignored. (E. g. when printing a string, all characters except first are printed).

If I don't use SD card, the diagram looks as follows:
[image]
As you can see here, **SCK polarity is initially 1 (mode 3), and so the SPI transaction works correctly.**

# Possible explanation

It seems that `spiTransaction` does not really change SPI mode immediately, but only after some writes start happening.

The offending code is https://github.com/moononournation/Arduino_GFX/blob/5258ba9dd3423127c4eeb5b2c83ab15c5a7014ae/src/databus/Arduino_ESP32SPI.cpp#L331. Here's how I was able to fix it:

Here's how I was able to fix this:
```diff
diff --git a/src/databus/Arduino_ESP32SPI.cpp b/src/databus/Arduino_ESP32SPI.cpp
index 32040f4..f5d71f8 100644
--- a/src/databus/Arduino_ESP32SPI.cpp
+++ b/src/databus/Arduino_ESP32SPI.cpp
@@ -342,6 +342,18 @@ void Arduino_ESP32SPI::beginWrite()
   {
     DC_HIGH();
   }
+
+  // Write dummy byte to ensure CPOL/CPHA enter correct state (in case previous SPI mode was different)
+  if (_dc == GFX_NOT_DEFINED) // 9-bit SPI
+  {
+    WRITE9BIT(0);
+  }
+  else
+  {
+    WRITE8BIT(0);
+  }
+  flush_data_buf();
+
   CS_LOW();
 }
```

Basically, this code writes a dummy byte into SPI while the screen is still deactivated. This fixes clock polarity, and all subsequent writes work properly. Here's the diagram after the fix:
[image]

Now I'm able to use same SPI for both SD card and my TFT display. Let me know if I should create a PR.

moononournation, 2024-02-06:
Can you help to test is it have same problem if you use HWSPI class?

and3rson, 2024-02-06:
@moononournation sure, give me few mins

and3rson, 2024-02-06:
Yes, same issue:
[image]
You can see final part of uSD transaction, followed by a brief ~0.2ms pause, followed by SCK staying low even after /LCD is asserted (should have named it /TFT... :) )

moononournation, 2024-02-07:
Oh, you mean you passed not defined in MISO parameter while declaring SPI class? Try pass the value shared with the SD card.

and3rson, 2024-02-07:
I created a relevan issue in ESP32 Arduino repo: https://github.com/espressif/arduino-esp32/issues/9221

and3rson, 2024-02-07:
@moononournation FWIW, for now I think the most reliable solution is simply to write a dummy byte before asserting /CS (as I described in my first message). What do you think?

moononournation, 2024-02-08:
I think write(0) is reasonable, what is the criteria need this trick? changed SPI mode? or always?

and3rson, 2024-02-08:
@moononournation Yeah, it would make sense to only write(0) if mode was different before `beginTransaction` was called.
Something like the following pseudo-code:

```cpp
uint8_t mode = _spi->mode;
beginTransaction(...);
if (mode != _spi->mode) {
    write(0);
}
```

moononournation, 2024-02-08:
Seems not easy to check current mode, now will write(0) if _is_shared_interface.

and3rson, 2024-02-08:
> Seems not easy to check current mode, now will write(0) if _is_shared_interface.

Sounds reasonable. I agree that 1-byte overhead does not seem like a problem.

and3rson, 2024-02-08:
@moononournation I think you also need to call `flush_data_buf()` after write?

and3rson, 2024-02-08:
Just tested your fix - everything is working properly! uSD & TFT are now nicely sharing the SPI bus.

Thanks again!

closed 2024-02-08 by and3rson

and3rson, 2024-02-17:
Hi @moononournation!
Sorry for bothering, but would you mind releasing a new version with this fix? We're currently using git version and would prefer to pin versions of our dependencies, since there will be lots of people using our framework soon, so I was wondering if 1.4.5 could happen anytime soon?
Thanks in advance!

moononournation, 2024-02-23:
should be released

moononournation, 2024-03-19:
ok, revoked changes.