moononournation/Arduino_GFX #436: print() doesn't work with custom GFX class

Spongman, 2024-02-12:
v1.4.4 on rp2040, arduino, platformio.

```
void setup()
{
	class MyGFX : public Arduino_GFX
	{
		std::string foo;

	public:
		MyGFX(int16_t w, int16_t h) : Arduino_GFX(w, h)
		{
		}

		virtual bool begin(int32_t speed = GFX_NOT_DEFINED) { return true; }

		virtual void writePixelPreclipped(int16_t x, int16_t y, uint16_t color)
		{
			Serial.printf("writePixelPreclipped %i, %i, %x\n", x, y, color);
		}
	};

	MyGFX buffer(240, 135);

	Serial.println(F("start"));
	buffer.startWrite();

	buffer.drawPixel(0, 0, RGB565(255, 255, 0));

	buffer.setCursor(10, 10);
	buffer.setTextColor(RGB565_GREEN);
	buffer.print("text");

	buffer.endWrite();
	Serial.println(F("done"));
}
```

for some reason, `buffer.print()` isn't doing anything, and I can't tell why. all i get is
```
start
writePixelPreclipped 0, 0, ffe0
done
```

If I comment out the `std::string foo;` line, then it works:
```
start
writePixelPreclipped 0, 0, ffe0
writePixelPreclipped 10, 12, 7e0
writePixelPreclipped 11, 12, 7e0
writePixelPreclipped 12, 10, 7e0
writePixelPreclipped 12, 11, 7e0
writePixelPreclipped 12, 12, 7e0
...
```

moononournation, 2024-02-13:
you did not mention your method should override parent class

Spongman, 2024-02-13:
not sure what you mean. the `writePixelPreclipped()` method above _does_ override the one in `Arduino_GFX`

moononournation, 2024-02-13:
you may refer other class how can override it

Spongman, 2024-02-13:
can you suggest a fix?

moononournation, 2024-02-13:
no idea

Spongman, 2024-02-13:
ok, so this 

> you did not mention your method should override parent class

isn't the cause ?