moononournation/Arduino_GFX #438: print() doesn't work with custom GFX class

Spongman, 2024-02-13:
going to try again, since the previous issue was wrongly closed:

v1.4.4 on rp2040, arduino, platformio.

```
#include <Arduino_GFX_Library.h>

void setup()
{
	class MyGFX : public Arduino_GFX
	{
		int foo;

	public:
		MyGFX(int16_t w, int16_t h) : Arduino_GFX(w, h)
		{
		}

		bool begin(int32_t speed = GFX_NOT_DEFINED) override { return true; }

		void writePixelPreclipped(int16_t x, int16_t y, uint16_t color) override
		{
			Serial.printf("writePixelPreclipped %i, %i, %x\n", x, y, color);
		}
	};

	MyGFX buffer(240, 135);

	Serial.println(F("start"));
	buffer.startWrite();

	buffer.drawPixel(0, 0, RGB565(255, 255, 0));

	buffer.setCursor(10, 10);
	buffer.setTextColor(RGB565_GREEN);
	buffer.print("text");

	buffer.endWrite();
	Serial.println(F("done"));
}
```

for some reason, buffer.print() isn't doing anything, and I can't tell why. all i get is

````
start
writePixelPreclipped 0, 0, ffe0
done
```

If I comment out the int foo; line, then it works:

```
start
writePixelPreclipped 0, 0, ffe0
writePixelPreclipped 10, 12, 7e0
writePixelPreclipped 11, 12, 7e0
writePixelPreclipped 12, 10, 7e0
writePixelPreclipped 12, 11, 7e0
writePixelPreclipped 12, 12, 7e0
...

Spongman, 2024-02-13:
ok, the cause of the bug (that has nothing to do with the `override` keyword, btw) is that the `_(min|max)_text_(x|y)` fields are *uninitialized*, and have predictable but undefined values. the commenting/uncommenting of the line of code was a red-herring, it just resulted in different codegen and a different set of undefined values for the uninitialized fields.

moononournation, 2024-02-22:
How to implement a new class is a discussion, so it converted to discussion. Please do not try to raise duplicate discussion in issue again.

closed 2024-02-22 by moononournation

Spongman, 2024-02-26:
this bug has nothing to do with "implement a new class". the bug is that the min/max text fields are uninitialized.

please take care to read and understand issues that people raise before closing them out of hand.

it seems very strange to me that you're unable to acknowledge the true cause of this issue, and persist in telling me that I don't know how to code c++, when clearly it is _you_ that is mistaken about how the `override` keyword works.

moononournation, 2024-03-03:
If it is a bug, the issue should report all child class print() doesn't work, right?
The override keyword just my previous experience about the unexpected behavior of some child classes. As you may already found, the hobbyist level coder may not give you the answer as accurate as ChatGPT. Sorry for I cannot be one of your guider in your coding journey. As mentioned in the library main page, I have very limited time that only can focus on the actual bugs the public suffering and my preferred enhancements.