moononournation/Arduino_GFX #444: Versions newer than 1.3.5 do not compile for ESP32 in PlatformIO with Arduino framework

JensKolb, 2024-02-20:
Hi,
I'm new to PlatformIO so I might be missing something, but a clean project for ESP32 on Arduino framework does not compile when I add this library with any version newer than 1.3.5 with the following error:

```
.pio\libdeps\esp32dev\GFX Library for Arduino\src\databus\Arduino_ESP32QSPI.cpp: In member function 'virtual bool Arduino_ESP32QSPI::begin(int32_t, int8_t)':
.pio\libdeps\esp32dev\GFX Library for Arduino\src\databus\Arduino_ESP32QSPI.cpp:58:43: error: 'SPICOMMON_BUSFLAG_GPIO_PINS' was not declared in this scope
       .flags = SPICOMMON_BUSFLAG_MASTER | SPICOMMON_BUSFLAG_GPIO_PINS,
                                           ^
```

Here is the full log:
<details>
<summary>Error Log</summary>

```
Processing esp32dev (platform: espressif32; board: nodemcu-32s; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (3.5.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6)
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- GFX Library for Arduino @ 1.4.4
|-- SPI @ 1.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\lib9d9\SPI\SPI.cpp.o
Compiling .pio\build\esp32dev\libdc7\Wire\Wire.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\Arduino_DataBus.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\Arduino_G.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\Arduino_GFX.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\Arduino_GFX_Library.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\Arduino_TFT.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\Arduino_TFT_18bit.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\canvas\Arduino_Canvas.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\canvas\Arduino_Canvas_3bit.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\canvas\Arduino_Canvas_Indexed.cpp.o
Archiving .pio\build\esp32dev\lib9d9\libSPI.a
Indexing .pio\build\esp32dev\lib9d9\libSPI.a
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\canvas\Arduino_Canvas_Mono.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_AVRPAR16.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_AVRPAR8.cpp.o
Archiving .pio\build\esp32dev\libdc7\libWire.a
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_DUEPAR16.cpp.o
Indexing .pio\build\esp32dev\libdc7\libWire.a
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32LCD16.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32LCD8.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32PAR16.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32PAR16Q.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32PAR16QQ.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32PAR8.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32PAR8Q.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32PAR8QQ.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32PAR8QQQ.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32QSPI.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32RGBPanel.cpp.o
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32S2PAR16.cpp.o
.pio\libdeps\esp32dev\GFX Library for Arduino\src\databus\Arduino_ESP32QSPI.cpp: In member function 'virtual bool Arduino_ESP32QSPI::begin(int32_t, int8_t)':
.pio\libdeps\esp32dev\GFX Library for Arduino\src\databus\Arduino_ESP32QSPI.cpp:58:43: error: 'SPICOMMON_BUSFLAG_GPIO_PINS' was not declared in this scope
       .flags = SPICOMMON_BUSFLAG_MASTER | SPICOMMON_BUSFLAG_GPIO_PINS,
                                           ^
[... 83 lines cut ...]
.pio\libdeps\esp32dev\GFX Library for Arduino\src\databus\Arduino_ESP32QSPI.cpp: In member function 'virtual void Arduino_ESP32QSPI::writeIndexedPixels(uint8_t*, uint16_t*, uint32_t)':
.pio\libdeps\esp32dev\GFX Library for Arduino\src\databus\Arduino_ESP32QSPI.cpp:544:60: error: 'SPI_TRANS_VARIABLE_DUMMY' was not declared in this scope
                                  SPI_TRANS_VARIABLE_ADDR | SPI_TRANS_VARIABLE_DUMMY;
                                                            ^
.pio\libdeps\esp32dev\GFX Library for Arduino\src\databus\Arduino_ESP32QSPI.cpp: In member function 'virtual void Arduino_ESP32QSPI::writeIndexedPixelsDouble(uint8_t*, uint16_t*, uint32_t)':
.pio\libdeps\esp32dev\GFX Library for Arduino\src\databus\Arduino_ESP32QSPI.cpp:597:60: error: 'SPI_TRANS_VARIABLE_DUMMY' was not declared in this scope
                                  SPI_TRANS_VARIABLE_ADDR | SPI_TRANS_VARIABLE_DUMMY;
                                                            ^
Compiling .pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32S2PAR16Q.cpp.o
*** [.pio\build\esp32dev\lib757\GFX Library for Arduino\databus\Arduino_ESP32QSPI.cpp.o] Error 1
```

</details>

It seems like it tries to use definitions of ESP-IDF which are not available when using the Arduino framework, but that's just speculation on my end.

This is how the problematic platform.ini looks like:
```
[env:esp32dev]
platform = espressif32
board = nodemcu-32s
framework = arduino
lib_deps = 
	moononournation/GFX Library for Arduino@^1.4.4
```

moononournation, 2024-02-20:
Arduino_GFX only support Arduino IDE

closed 2024-02-22 by moononournation

moononournation, 2024-02-24:
I am not using PIO so cannot support it. And if the problem not exists in Arduino IDE, I think the right way is report this to PIO.