moononournation/Arduino_GFX #445: Underscore character issue.

jobmathew, 2024-02-21:
Hi, I am using Arduino_GC9A01 with 128*128 round display. Everything's working fine except underscore character (ascii 95). Not properly lineup strings when there is underscore character. This will display on next line always. I am using SanBold12pt font.

jobmathew, 2024-02-22:
Font working fine. fixed the problem. Thank you.
But I have another issue happened when latest library installed. My icon colors changed blue to red and red to blue.

jobmathew, 2024-02-22:
Icon color is ok on version 1.4.4. Latest version1.4.5 is the issues.

moononournation, 2024-02-22:
this version changed the behavior if
isSharedInterface = true
you may set it to false to check it.

moononournation, 2024-02-23:
Any compare photo?

jobmathew, 2024-02-24:
I don’t have photos.

closed 2024-03-03 by moononournation