moononournation/Arduino_GFX #449: 16 bit i8080 interface  ESP32s3+st7796 code，doesn't display anything

774471872, 2024-03-07:
Has the code of the 16 bit i8080 interface driver for the 320x480 screen of ESP32s3+st7796 been tested? Why did I run the example/PDQgraphicstest routine as instructed, but the screen still doesn't display anything

moononournation, 2024-03-07:
any details?

774471872, 2024-03-08:
[details.pptx](https://github.com/moononournation/Arduino_GFX/files/14532468/details.pptx)
[arduino_gfx.zip](https://github.com/moononournation/Arduino_GFX/files/14532472/arduino_gfx.zip)
This is the PowerPoint I made, which includes configuration details; My source code file has also been uploaded

moononournation, 2024-03-08:
please use the format that can view in github

774471872, 2024-03-08:
[image]
This is the configuration of Arduino_GFX_databus. h. My SOC is esp32s3, 
and the communication method with the monitor is a 16 bit i8080 interface

[image]
My device is esp32-s3-devkitc-1, and there is no device in Arduino_GFX_dev_device. h, 
so I did not make any modifications

[image]
I uncommented this code in Arduino_GFX_display. h,
Because my monitor driver chip is ST7796s and the screen size is 320 * 480

[image]
I did not make any changes in arduino_GFX_pins. h 
because the development board I selected in arduinoIDE is esp32s3 dev module

774471872, 2024-03-08:
[image]
This is a photo of the hardware
1. That lead resistor is the pull-up resistor of the RD pin, but adding this resistor doesn't seem to be useful
2. As shown in the figure, the screen backlight is normal, but the content is not displayed, only a white patch

moononournation, 2024-04-12:
not tested, seems not working

closed 2024-04-17 by moononournation