moononournation/Arduino_GFX #452: RP2040 1.28 WAVESHARE TOUCH LCD

tonluc123, 2024-03-10:
Hello,
As I had no answers at the discussions I thought it would be better to post my issue here
I'm using this library successfully for almost a year now. I used GC9A01 & GC9A01 with touch and it is great. Lately I purchased a RP2040 1.28 WAVESHARE TOUCH LCD which I thought it would be easy to manage as it uses the same hardware , but ... no luck.
I tried the clock example and used the Dev-Device-Declaration#waveshare-rp2040-lcd-128 but nothing comes on the screen.
Does anybody had any experience with this module? Any help would be appreciated.
Lucas
Code her:
#include <Arduino_GFX_Library.h>
#define GFX_BL 25
Arduino_DataBus *bus = new Arduino_RPiPicoSPI(8 /* DC */, 9 /* CS */, 10 /* SCK */, 11 /* MOSI */, 12 /* MISO */, spi1 /* spi */);
Arduino_GFX *gfx = new Arduino_GC9A01(bus, 12 /* RST */, 0 /* rotation */, true /* IPS */);

void setup() {
 Serial.begin(115200);
  gfx->begin();
  gfx->fillScreen(BLUE);
}

void loop() {
  // put your main code here, to run repeatedly:

}

closed 2024-03-10 by moononournation

tonluc123, 2024-03-10:
I tried , no success. Is it possible to send me the code tat works for you ? I will find what I did wrong by comparing. Also, I noticed that in the background you use ide 1.8xxx, in my case it is 2.3.0.
Lucas

moononournation, 2024-03-10:
already show you PDQgraphicstest works. no code change, just select correct board

tonluc123, 2024-03-10:
I'm really sorry I'm wasting your time but I have no other help. As you may see from the photo attached the board is the correct one and the serial monitor indicates that the program is running. I just do not have anything in my screen.
[image]