moononournation/Arduino_GFX #454: Compilation error for BW16-kit

KilimSerg, 2024-03-17:
Hello moononournation!

I don't speak English so I apologize in advance for any inaccuracies in Google Translate
I'm trying to replicate the “RTLWiFiAnalyzer” project on the BW16-kit (RTL8720DN) board.
 
[image]

and encountered a problem when compiling a program that uses the “Arduino GFX library”.
If there are no logical or syntax errors at the end of the compilation, the Arduino IDE throws an error that it cannot compile the program for a specific board

Arduino IDE 1.8.13
exit status 1
Error compiling for board Ai-Thinker BW16

Arduino IDE 2.2.1
exit status 1
Compilation error: exit status 1

As far as I understand from the compiler message, the problem lies in two libraries
“Arduino GFX library” and “lwip_netconf.h”.
I tried to run the “PDQgraphicstest.ino” test from the “Arduino GFX library” for the ILI9341 display and no matter how I tried to configure it on the Ai-Thinker BW16 board I needed, the compiler still gave the same error.

exit status 1
Error compiling for board Ai-Thinker BW16 (RTL8720DN).

Since the list of Ai-Thinker BW16 boards is not in the example, I tried to independently configure the necessary pins of my board

#include <Arduino_GFX_Library.h>
Arduino_DataBus *bus = new Arduino_SWSPI(8 /* DC */, 9 /* CS */, 10 /* SCK */, 12 /* MOSI */, 11 /* MISO */);
Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 3 /* rotation */, false /* IPS */);

but still got the same error.
I tried to compile the example from amebe “ScanNetworks_LCD.ino” and everything turned out fine. The example works fine on the display. I give an example below, although it does not use the “lwip_netconf.h” library.
 
[image]

I am not an experienced programmer and it is difficult for me to determine the cause of the errors that I receive.
I would like to ask you to help me figure out what is the cause of the error when using the “Arduino GFX library”.
And if you have such an opportunity, what is the reason for the error to appear due to the “lwip_netconf.h” library.

Below is a listing of the program compilation:

In file included from c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:4:0,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024217-7288-19klnw8.xxp9\RTLWiFiAnalyzer\RTLWiFiAnalyzer.ino:27:
c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_DataBus.h:145:6: warning: "ESP_ARDUINO_VERSION_MAJOR" is not defined, evaluates to 0 [-Wundef]
 #if (ESP_ARDUINO_VERSION_MAJOR < 3)
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:9:0,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024217-7288-19klnw8.xxp9\RTLWiFiAnalyzer\RTLWiFiAnalyzer.ino:27:
c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32LCD8.h:5:6: warning: "ESP_ARDUINO_VERSION_MAJOR" is not defined, evaluates to 0 [-Wundef]
 #if (ESP_ARDUINO_VERSION_MAJOR < 3)
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:10:0,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024217-7288-19klnw8.xxp9\RTLWiFiAnalyzer\RTLWiFiAnalyzer.ino:27:
c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32LCD16.h:7:6: warning: "ESP_ARDUINO_VERSION_MAJOR" is not defined, evaluates to 0 [-Wundef]
 #if (ESP_ARDUINO_VERSION_MAJOR < 3)
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:19:0,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024217-7288-19klnw8.xxp9\RTLWiFiAnalyzer\RTLWiFiAnalyzer.ino:27:
c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:3:6: warning: "ESP_ARDUINO_VERSION_MAJOR" is not defined, evaluates to 0 [-Wundef]
 #if (ESP_ARDUINO_VERSION_MAJOR < 3)
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:81:0,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024217-7288-19klnw8.xxp9\RTLWiFiAnalyzer\RTLWiFiAnalyzer.ino:27:
c:\Users\User\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/display/Arduino_RGB_Display.h:3:6: warning: "ESP_ARDUINO_VERSION_MAJOR" is not defined, evaluates to 0 [-Wundef]
 #if (ESP_ARDUINO_VERSION_MAJOR < 3)
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/network/lwip/lwip_v2.0.2/src/include/lwip/timeouts.h:44:0,
                 from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/network/lwip/lwip_v2.0.2/src/include/lwip/tcpip.h:45,
                 from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/api/lwip_netconf.h:18,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024217-7288-19klnw8.xxp9\RTLWiFiAnalyzer\RTLWiFiAnalyzer.ino:20:
C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/network/lwip/lwip_v2.0.2/src/include/lwip/sys.h:351:7: error: conflicting declaration of 'u32_t sys_now()' with 'C' linkage
 u32_t sys_now(void);
       ^~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/network/lwip/lwip_v2.0.2/src/include/lwip/opt.h:51:0,
                 from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/network/lwip/lwip_v2.0.2/src/include/lwip/errno.h:40,
                 from c:\users\user\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\cerrno:42,
                 from c:\users\user\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\ext\string_conversions.h:44,
                 from c:\users\user\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\basic_string.h:5429,
                 from c:\users\user\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\string:52,
                 from C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\cores\ambd/Arduino.h:36,
                 from C:\Users\User\AppData\Local\Temp\arduino\sketches\7639B3E19B93B90426C91C194E6DEA76\sketch\RTLWiFiAnalyzer.ino.cpp:1:
C:\Users\User\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\system/component/common/api/network/include/lwipopts.h:177:21: note: previous declaration with 'C++' linkage
 extern unsigned int sys_now(void);
                     ^~~~~~~

exit status 1

Compilation error: exit status 1

Arduino IDE 1.8.13
exit status 1
Error compiling for board Ai-Thinker BW16 (RTL8720DN).

Best regards, Sergei.

P.S.
I have another question for you that is a little off topic.
Have you come across documentation on programming the RTL8720DN chip at the level of accessing directly to the register of the chip, for example, to manually set the initial scanning frequency, the final frequency, read the signal level at a specific frequency, turn on radiation at a specific frequency, and so on.
From what I looked on the Internet, I didn’t come across anything similar to this microcircuit, maybe you have?
The “nrf2401” module can be programmed in a similar way.

moononournation, 2024-03-19:
please get update the latest master branch to check is it ok now

KilimSerg, 2024-03-19:
Hello moononournation!

I updated the library as you recommended.
And I tried to compile the simplest example from the “Arduino GFX library”
  "HelloWorld.ino" and got 2 errors.

I’ll give you a complete example so you can see, maybe I’m doing something wrong.
I set the pins of my board according to the example you give for the RTL8720DN
on the page: https://www.instructables.com/ArduinoGFX/

Connection summary:
RTL8720DN 		-> ILI9341
Vin			 -> Vcc
GND			 -> GND
GPIO 18		 -> CS
GPIO 2 		-> RESET
GPIO 17 		-> D/C
GPIO 21(MOSI) 	-> SDI(MOSI)
GPIO 19(SCK) 	-> SCK
GPIO 23 		-> LED

Sketch:
/*******************************************************************************
 * Start of Arduino_GFX setting
 *
 * Arduino_GFX try to find the settings depends on selected board in Arduino IDE
 * Or you can define the display dev kit not in the board list
 * Defalult pin list for non display dev kit:
 * Arduino Nano, Micro and more: CS:  9, DC:  8, RST:  7, BL:  6, SCK: 13, MOSI: 11, MISO: 12
 * ESP32 various dev board     : CS:  5, DC: 27, RST: 33, BL: 22, SCK: 18, MOSI: 23, MISO: nil
 * ESP32-C3 various dev board  : CS:  7, DC:  2, RST:  1, BL:  3, SCK:  4, MOSI:  6, MISO: nil
 * ESP32-S2 various dev board  : CS: 34, DC: 38, RST: 33, BL: 21, SCK: 36, MOSI: 35, MISO: nil
 * ESP32-S3 various dev board  : CS: 40, DC: 41, RST: 42, BL: 48, SCK: 36, MOSI: 35, MISO: nil
 * ESP8266 various dev board   : CS: 15, DC:  4, RST:  2, BL:  5, SCK: 14, MOSI: 13, MISO: 12
 * Raspberry Pi Pico dev board : CS: 17, DC: 27, RST: 26, BL: 28, SCK: 18, MOSI: 19, MISO: 16
 * RTL8720 BW16 old patch core : CS: 18, DC: 17, RST:  2, BL: 23, SCK: 19, MOSI: 21, MISO: 20
 * RTL8720_BW16 Official core  : CS:  9, DC:  8, RST:  6, BL:  3, SCK: 10, MOSI: 12, MISO: 11
 * RTL8722 dev board           : CS: 18, DC: 17, RST: 22, BL: 23, SCK: 13, MOSI: 11, MISO: 12
 * RTL8722_mini dev board      : CS: 12, DC: 14, RST: 15, BL: 13, SCK: 11, MOSI:  9, MISO: 10
 * Seeeduino XIAO dev board    : CS:  3, DC:  2, RST:  1, BL:  0, SCK:  8, MOSI: 10, MISO:  9
 * Teensy 4.1 dev board        : CS: 39, DC: 41, RST: 40, BL: 22, SCK: 13, MOSI: 11, MISO: 12
 ******************************************************************************/
#include <Arduino_GFX_Library.h>

#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_DataBus *bus = new Arduino_HWSPI(8 /* DC */, 9 /* CS */);

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
Arduino_GFX *gfx = new Arduino_ILI9341(bus, 6, 0 /* rotation */, false /* IPS */);

#endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{
  Serial.begin(115200);
  // Serial.setDebugOutput(true);
  // while(!Serial);
  Serial.println("Arduino_GFX Hello World example");

#ifdef GFX_EXTRA_PRE_INIT
  GFX_EXTRA_PRE_INIT();
#endif

  // Init Display
  if (!gfx->begin())
  {
    Serial.println("gfx->begin() failed!");
  }
  gfx->fillScreen(BLACK);

#ifdef GFX_BL
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);
#endif

  gfx->setCursor(10, 10);
  gfx->setTextColor(RED);
  gfx->println("Hello World!");

  delay(5000); // 5 seconds
}

void loop()
{
  gfx->setCursor(random(gfx->width()), random(gfx->height()));
  gfx->setTextColor(random(0xffff), random(0xffff));
  gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
  gfx->println("Hello World!");

  delay(1000); // 1 second
}

But still, even with such a simple example, I get two errors !!!!!

C:\Program Files (x86)\Arduino\libraries\Arduino_GFX-master\src\databus\Arduino_HWSPI.cpp: In member function 'virtual bool Arduino_HWSPI::begin(int32_t, int8_t)':
C:\Program Files (x86)\Arduino\libraries\Arduino_GFX-master\src\databus\Arduino_HWSPI.cpp:184:17: error: 'SPI_MODE2' was not declared in this scope
     _dataMode = SPI_MODE2;
                 ^~~~~~~~~
exit status 1
Error compiling for board Ai-Thinker BW16 (RTL8720DN).

And I must say that I looked through the “Arduino_HWSPI.cpp” file, I did not find those lines that are indicated in the errors “Arduino_HWSPI::begin(int32_t, int8_t)” and “SPI_MODE2”.

Best regards, Sergei.

moononournation, 2024-03-19:
I think you should report to realtek not defined SPI_MODE2

KilimSerg, 2024-03-20:
But there are many examples on the Internet of repeating the wi-fi analyzer on the RTL8720DN, and you too, where everything works fine, how did the wi-fi analyzer work normally before?

moononournation, 2024-03-20:
that's why I suggest you report this error to Realtek latest board package have this error.

KilimSerg, 2024-03-20:
Okay, I’ll try to get in touch, but I have a few more questions for you:

You can indicate a specific address where to contact, I received the BW16-kit (RTL8720DN) boards only 2 weeks ago and did not expect to encounter such problems when trying to repeat the finished ones. proven and working examples from the Internet.
Therefore, I simply don’t know where to send my request.
And the second question.
Maybe you can tell me where I can get older software that did not have this error yet.
Since the request itself and its corrections may take some time, and it is not known how long, I would not want to sit and just wait.

moononournation, 2024-03-20:
use previous version is a quick solution, you can use board manager select previous version to install

KilimSerg, 2024-03-20:
Can you tell me the address where to send a bug report?

KilimSerg, 2024-03-21:
Hello moononournation!
Thank you for your advice, rolling back to an older version really helped.
True, the rollback was not without problems.
I first tried to roll back through the Arduino IDE board manager and the problem with “Arduino GFX Library” went away, but the problem with “lwip_netconf.h” remained, I tried to roll back to a lower version again, after that the “Arduino GFX Library” stopped working altogether.
After several reinstallations, I finally figured out that I need to completely delete the “realtek” folder (C:\Users\User\AppData\Local\Arduino15\packages\realtek\) and only after that install the old version through the board manager, as if you were installing it the first time and then everything worked great. It is enough to roll back to version 3.1.6.
I am attaching a photo:

[image]
[image]

moononournation, 2024-03-22:
Good to see that.

You may report the Ameba latest release issue at their repository:
https://github.com/ambiot/ambd_arduino

closed 2024-03-22 by moononournation