moononournation/Arduino_GFX #455: Waveshare display 3.5inches

pdo59, 2024-03-20:
Hi,
I'm currently working on a project that utilizes the [3.5" TFT Touch Shield from Waveshare](https://www.waveshare.com/wiki/3.5inch_TFT_Touch_Shield), but I've encountered an issue when trying to use it with your Arduino GFX library. While the display operates flawlessly with the original example code provided by Waveshare (found [here](https://www.waveshare.com/wiki/3.5inch_TFT_Touch_Shield)), it does not seem to work when I attempt to integrate it with your library.
Here are the specific settings I used in my attempt:

`#include <Arduino_GFX_Library.h>

#define GFX_BL 9 // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

#include <Arduino_GFX_Library.h>

Arduino_DataBus *bus = new Arduino_HWSPI(7 /* DC */, 10 /* CS */);

Arduino_GFX *gfx = new Arduino_ILI9486( bus, 8 /* RST */, 0 /* rotation */, false /* IPS */);
`

I suspect there might be a compatibility issue or a specific setting in your library that I'm missing. Could you please offer any guidance on how to resolve this issue? Any advice or recommendations would be greatly appreciated.
Thank you for your time and assistance.
Best regards

moononournation, 2024-03-21:
SPI + ILI948x combination is a little tricky, because did not support 16-bit color. You should use Arduino_ILI9486_18bit

pdo59, 2024-03-21:
Hi, Thank you very much for your prompt response and suggestion.  I have already tried using the Arduino_ILI9486_18bit constructor in my attempts, as indicated in my initial message. Unfortunately, I still haven't been able to get the display to work correctly with this setup.
```
#define GFX_BL 9 // default backlight pin, you may replace DF_GFX_BL to actual backlight pin
Arduino_DataBus *bus = new Arduino_HWSPI(7 /* DC */, 10 /* CS */);
Arduino_GFX *gfx = new Arduino_ILI9486_18bit( bus, 8 /* RST */, 0 /* rotation */, false /* IPS */);
```
I used this connections: 
```
SCLK | D13
MISO | D12
MOSI | D11
LCD_CS | D10
LCD_BL | D9
LCD_RST | D8
LCD_DC | D7
TP_BUSY | D6
SD_CS | D5
TP_CS | D4
TP_IRQ | D3
```
Given this, I'm wondering if there might be another aspect of the configuration or setup that I'm missing, or if there's another workaround or step that I should try to resolve this issue.
Any further insights or recommendations you could offer would be greatly appreciated.
Thanks again for your time and support.

moononournation, 2024-03-21:
I found the SPI shared with SD and TP, you may need to pullup both CS to avoid signal conflict.

pdo59, 2024-03-21:
To ensure I'm understanding your suggestion correctly, are you recommending that I add pull-up resistors to the CS signals of both the SD card and the TP?

moononournation, 2024-03-21:
you can use pinMode and digitalWrite high to pull up in software

pdo59, 2024-03-21:
nothing to do, it doesn't work

moononournation, 2024-03-21:
thats all I know

pdo59, 2024-03-21:
Thank you again for your support. 
I'll continue to investigate and explore other potential fixes or workarounds for the problem. If anyone else in the community has experienced something similar or has suggestions, I would be grateful for any additional advice.

moononournation, 2024-03-22:
I found the waveshare init code using 16-bit color, but ILI9486 datasheet did not mention it. And I tried other ILI9486 display cannot use 16-bit, so I believe it actually ST7796. you may try Arduino_ST7796 is it work.

closed 2024-03-22 by moononournation

pdo59, 2024-03-22:
Hi, @moononournation,
I tested HelloWorld.ino example with following settings: 
```
#define GFX_BL 9//DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

Arduino_DataBus *bus = new Arduino_HWSPI(7, 10);
Arduino_GFX *gfx = new Arduino_ST7796(bus, 8 /*RST*/ , 0 /* rotation */);
```
but it doesn't work, the backlight comes on and nothing else; I also tried with IPS = true.