moononournation/Arduino_GFX #458: Memory free about SpriteGif example

Oopas, 2024-04-01:
I modified the example `Sprite/SpriteGif/SpriteGif.ino` to read the exit drawing function when the button is pressed. Everything was fine when I didn't plan to release `spriteMaster` memory through the free function. But when I tried to add an `free(spriteMaster);` before the return, the program crashed.

How can I release the memory requested by `spriteMaster=(uint8_t *)malloc(Gif->width * Gif->height/2);`?

Here is the crash information:

CORRUPT HEAP: Bad tail at 0x3d832851. Expected 0xbaad5678 got 0xb1b1b1b1

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x403779aa:0x3fced070 0x4037e5b5:0x3fced090 0x40384f29:0x3fced0b0 0x40384b19:0x3fced1e0 0x40378089:0x3fced200 0x40384f59:0x3fced220 0x42007b0d:0x3fced240 0x42009ebc:0x3fced290 0x42025662:0x3fced2c0

ELF file SHA256: 0000000000000000

E (3495) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...

moononournation, 2024-04-01:
I have no idea how to direct free a class

Oopas, 2024-04-01:
> I have no idea how to direct free a class

I think I found the problem just now.....

`spriteMaster=(uint8_t *)malloc(Gif->width * Gif->height/2);` The requested memory size is the product of GIF image resolution/2, which is 405 * 407/2=82416. But it seems that 377 * 405 was used in the following `mpv = new IndexedSprite((LCD_Canvas->width() - 140) / 2, 182, spriteMaster + (377 * 405), palette, 50, 30, (405 - 50), false, 8, 2, Gif->gce.tindex);`.

This will be greater than the allocated memory, causing an out of bounds. After trying to remove `/2`, `free(spriteMaster);` successfully freed up memory.

closed 2024-04-12 by moononournation