moononournation/Arduino_GFX #464: No RM690B0 display driver for Lilygo T4-S3 (ESP32 Amoled Touch)

richgain, 2024-04-09:
Hi again,
I would love to be able to use Arduino_GFX with the Lilygo T-Display Amoled T4-S3 ( https://www.lilygo.cc/products/t4-s3 ) but have not yet been able to find a way to do so.

The LILYGO_T_DISPLAY_S3_AMOLED makes use of your Arduino_RM67162 driver and works very well, but the newer Amoled T4-S3 is based on the 450x600 pixel RM690B0 display and I have not found a suitable driver for this display yet.

Is this something that could be supported in future?
Thanks.

nikthefix, 2024-04-10:
@richgain 

The RM67162 driver is actually compatible with the T4-S3. You'll need to change the pin assignments and the QSPI frequency -  36MHz works for me. Also, the T4 display has a 16px positive offset on the vertical axis (in landscape mode). 

For the T4, gpio 9 needs to be set OUTPUT HIGH to enable the display.
Don't forget to change the display resolution to 600/450.

nik

richgain, 2024-04-10:
@nikthefix 
Many thanks, nik. That's really helpful information.

Richard

moononournation, 2024-05-12:
Latest version is supported LilyGo T4 S3

closed 2024-05-12 by moononournation