moononournation/Arduino_GFX #465: ESP32-4848S040C

Bzhmic, 2024-04-15:
Hi moonnourturation and thanks for your very good work.
I'm bored with my screen because it doesn't work well with the latest version of Arduino GFX Ver 1.4.6
my screen is an ESP32-4848S040C with an ST7701. https://homeding.github.io/boards/esp32s3/panel-4848S040.htm

it works well with Arduino GFX 129 with this config

#define GFX_BL 38 // BackLight 

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    39 /* CS    */, 48 /* SCK  */, 47 /* SDA */, 18 /* DE */, 17 /* VSYNC */,
    16 /* HSYNC */, 21 /* PCLK */,
    
    11 /* R0 */, 12 /* R1 */, 13 /* R2 */, 14 /* R3 */,  0 /* R4 */, 
     8 /* G0 */, 20 /* G1 */,  3 /* G2 */, 46 /* G3 */,  9 /* G4 */, 10 /* G5 */, 
     4 /* B0 */,  5 /* B1 */,  6 /* B2 */,  7 /* B3 */, 15 /* B4 */, false );

Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel(
    bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */, true /* IPS */, 480 /* width */, 480 /* height */,
    st7701_type1_init_operations, sizeof(st7701_type1_init_operations), true /* BGR */,
    10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
    10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */ );

[image]

mais pas avec Arduino GFX 146 et cette config 

#define GFX_BL 38 // BackLight
#define ESP32_S3_RGB

Arduino_DataBus *bus = new Arduino_SWSPI(
    GFX_NOT_DEFINED /* DC */, 39 /* CS */,   48 /* SCK */, 47 /* MOSI */, GFX_NOT_DEFINED /* MISO */);

// Uncomment for 4" rect display
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    18 /* DE */, 17 /* VSYNC */, 16 /* HSYNC */, 21 /* PCLK */,

  /*R*/   11 /* R0 */, 12 /* R1 */, 13 /* R2 */, 14 /* R3 */,  0 /* R4 */, 
  /*G*/    8 /* G0 */, 20 /* G1 */,  3 /* G2 */, 46 /* G3 */,  9 /* G4 */, 10 /* G5 */, 
  /*B*/    4 /* B0 */,  5 /* B1 */,  6 /* B2 */,  7 /* B3 */, 15 /* B4 */, 

    1 /* hsync_polarity */, 10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
    1 /* vsync_polarity */, 10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, st7701_type8_init_operations, sizeof(st7701_type8_init_operations));

part of the image at the bottom is missing
[image]

and with init1 the colors are inverted
[image]

when I have this white screen, I have to reset init 8 of v146 to have the right colors, because even with v129 the colors are inverted.

I use IDE 2.3.2, I have the same result with IDE 1.8.19

Have a good day

moononournation, 2024-04-17:
why you changed type1 to type8?

Bzhmic, 2024-04-17:
"and with init1 the colors are inverted"

Bzhmic, 2024-04-30:
The Guition 4848S040 work fine with the type 9 update.  
Thanks for the update Moononournation ;-)

closed 2024-05-12 by moononournation