moononournation/Arduino_GFX #472: JC8048W550 ST7262 no display with V1.4.6

hape65, 2024-05-04:
hi
thx for your work to get all the displays work. But unfortunately the described board above can be compiled with the PDQgraphicstest.ino but there is no picture on the screen.
Could i do a check on something else? The board has a screen programmed so it works. 
this are the pins:
Serial Number	Pin	 Notes  Connect	  
1	          IO8	 U1RXD	DB1(B)	b0            
2	          IO3	                DB2(B)	b1                
3	          IO46                      DB3(B)	b2              
4	          IO9	                DB4(B)	b3                
5	          IO1	                DB5(B)	b4                
6	          IO39                      HSYNC	  x                 
7	          IO41                      VSYNC	  x                 
8	          IO40                      DE	          x                   
9	          IO5	               DB6(G)	g0                
10	          IO7	               DB8(G)	g2              
11	          IO15                     DB9(G)	g3            
12	          IO16                     DB10(G)	g4            
13	          IO4	               DB11(G)	g5              
14	          IO45                     DB13(R)	r0            
15	          IO48                     DB14(R)	r1            
16	          IO47                     DB15(R)	r2            
17	          IO21                     DB16(R)	r3            
18	          IO43        U0TXD	TX	              
19	          IO44        U0RXD	RX	              
20	          IO0	  BOOT	I2S-BCLK/L3	        
21	          IO18        I2S-LRCLK/L2	        
22	          IO10   SPICS0	TF(CS)	        
23	          IO11   SPID	TF(MISO)	        
24	          IO17   U1TXD	IS2-DIN/L1	      
25	          IO38        TP_CS	                
26	          IO21        BL_C	                
27	          IO37        			                    
28	          IO36        			                    
29	          IO35        			                    
30	          IO14                  DB17(R)	r4            
31	          IO20   USB_D+	TP_SCL	        
32	          IO12   SPICLK	TF(SCK)/TP_CLK	
33	          IO13   SPIQ	TF(MOSI)	        
34	          IO42                       PCLK	  x               
35	          IO6	                 DB7(G)	g1              
36	          IO19   USB_D-	TP_SDA	        

so this same with this definition

// Uncomment for ST7262 IPS LCD 800x480
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
    0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

The output of the sketch:
Benchmark	micro-secs
Screen fill	164465
Text	20266
Pixels	1073943
Lines	522938
Horiz/Vert Lines	14731
Rectangles (filled)	342017
Rectangles (outline)	9770
Triangles (filled)	125138
Triangles (outline)	29756
Circles (filled)	58025
Circles (outline)	50625
Arcs (filled)	36282
Arcs (outline)	69928
Rounded rects (filled)	349063
Rounded rects (outline)	25338
Done!
Benchmark	micro-secs

Do you have any idea what i can test anymore?

regards
hape

moononournation, 2024-05-05:
BL?

moononournation, 2024-05-05:
is it the back light pin should be 21?

moononournation, 2024-05-05:
use arduino-esp32 v2

hape65, 2024-05-05:
ok one step ahead
now the sketch comes up at runtime with this:

Arduino_GFX PDQgraphicstest example!
E (6) lcd_panel.rgb: esp_lcd_new_rgb_panel(151): no mem for frame buffer
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4037ab7c
file: "c:\Users\hape\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32RGBPanel.cpp" line 119
func: uint16_t* Arduino_ESP32RGBPanel::getFrameBuffer(int16_t, int16_t)
expression: esp_lcd_new_rgb_panel(_panel_config, &_panel_handle)

abort() was called at PC 0x4037ab7f on core 1

Backtrace: 0x4037777e:0x3fcebad0 0x4037ab89:0x3fcebaf0 0x403807e1:0x3fcebb10 0x4037ab7f:0x3fcebb90 0x4200438b:0x3fcebbb0 0x420044fa:0x3fcebc00 0x42001c28:0x3fcebc20 0x4200731e:0x3fcebc50

moononournation, 2024-05-05:
please follow your ESP32 version set the flash size, speed and PSRAM type

hape65, 2024-05-05:
so finally it works thx a lot to you - is https://paypal.me/moononournation your paypal?

closed 2024-05-12 by moononournation