moononournation/Arduino_GFX #477: No image display from flash or SD (esp32S3-8048S070)

digitongithub, 2024-05-10:
I have a 7 inch display (esp32S3-8048S070), but I can't seem to show images from the flash or SD card on the display. I have the same display with LCD code and everything works there.
A listing of the SD card or flash just works and I see all the images
A standard Bitmap created with the code does work.
I'm trying to make a weather station with gfx code and then I want to leave the icons (bmp)

Have also tried the ImgViewerBmp program, but it also gives error messages.
(E:\Arduino\ImgViewerBmp\ImgViewerBmp.ino:111:18: error: could not convert 'gfx->Arduino_GFX::begin(-1)' from 'void' to 'bool')

moononournation, 2024-05-10:
I think you have multiple gfx folders in Arduino libraries. You may try delete all gfx folders and install the library again.

moononournation, 2024-05-10:
Many terms in your message is implicited other things, can clarify what you actually want to say?

digitongithub, 2024-05-10:
This with 7 inch display
Image in flash and placed in SD

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
Arduino_GFX BMP Image Viewer example
E (449) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1367:error: Corrupted dir pair at {0x0, 0x1}

E (449) esp_littlefs: mount failed,  (-84)
E (452) esp_littlefs: Failed to initialize LittleFS
ERROR: File System Mount Failed!

moononournation, 2024-05-10:
you not yet upload the littlefs files?

closed 2024-05-12 by moononournation

digitongithub, 2024-05-12:
I have all the images in the flash and on the SD card. When I use another program I also see the image in the flash and also on the SD card. But can't show these images on the display with any code.

digitongithub, 2024-05-15:
Here are the programs for the SD card and SPIFFS (with serial monitor).
[SD_serial.txt](https://github.com/moononournation/Arduino_GFX/files/15317769/SD_serial.txt)
[SD_LCD7.txt](https://github.com/moononournation/Arduino_GFX/files/15317770/SD_LCD7.txt)
[SPIFFS_Serial.txt](https://github.com/moononournation/Arduino_GFX/files/15317810/SPIFFS_Serial.txt)
[SPIFFS_LCD7.txt](https://github.com/moononournation/Arduino_GFX/files/15317811/SPIFFS_LCD7.txt)