moononournation/Arduino_GFX #478: ILI9488 with Arduino UNO R3

LimpidCrypto, 2024-05-21:
I'm new to programming with an Arduino. I'm using a Arduino UNO R3 and want to connect a [ILI9488 touch screen](https://amzn.eu/d/j3iyLgM) to it. Here is how I wired it:
- VCC: 5V
- GND: GND
- CS: 9
- RESET: 7
- D/C: 8
- SDI: 11
- SCK: 13
- LED: 5V
- SDO(MISO): 12
I just want to fill the screen blue for now. I use the same wiring in the [Wokwi Simulator](https://wokwi.com/) with a ILI9341 and it works fine. But with the ILI9488 I only the the nearly white screen with only these two stripes in the middle of the screen:
[image]
Here is my very simple code:
```
Arduino_DataBus *bus = create_default_Arduino_DataBus();
Arduino_ILI9488 *gfx = new Arduino_ILI9488(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);
gfx->begin();
gfx->fillScreen(RGB565_BLUE);
```
If it helps here is also the back of the screen:
[image]

If you need anything to be able to help me better, please tell me what you need. Thank you in advance.

moononournation, 2024-05-21:
if create_default_Arduino_DataBus() not work, you can declare databus yourself

LimpidCrypto, 2024-05-22:
I've tried the following but still no success. Am I using the right `DataBus`?
```cpp
Arduino_DataBus *bus = new Arduino_HWSPI(8 /* DC */, 9 /* CS */);
```

moononournation, 2024-05-22:
if you use SPI, you need to use Arduino_ILI9488_18bit

LimpidCrypto, 2024-05-22:
I now have tried using the `Arduino_ILI9488_18bit` and the display gets completely blue as soon as it resets. Once it starts again it's just as before with those two stripes and just a little bit more blue. This is my code now:
```cpp
Arduino_DataBus *bus = new Arduino_HWSPI(8 /* DC */, 9 /* CS */);
  Arduino_GFX *gfx = new Arduino_ILI9488_18bit(bus, 7 /* RST */, 1 /* rotation */, false /* IPS */);
  gfx->begin();
  gfx->fillScreen(RGB565_BLUE);
```

moononournation, 2024-05-22:
a good forward

LimpidCrypto, 2024-05-22:
> a good forward

Thank you for your help. Do you have an idea how to fix this issue?

moononournation, 2024-05-22:
check wire contact one by one

closed 2024-06-02 by moononournation