moononournation/Arduino_GFX #481: vscode+platformio ESP32S3 error

mspudi, 2024-05-25:
Arduino IDE ESP32S3   Display normal   
/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_DataBus *bus = new Arduino_ESP32SPI(41 /* DC */, 40 /* CS */, 36 /* SCK */, 35 /* MOSI */, GFX_NOT_DEFINED /* MISO */, FSPI /* spi_num */);

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
Arduino_GFX *gfx = new Arduino_HX8357A(bus, 42 /* RST */, 0 /* rotation */, true /* IPS */);

 vscode+platformio ESP32S3  board = esp32-s3-devkitc-1 board_build.mcu = esp32s3  platform = espressif32 @ 6.3.2

Serial port printing error

：uru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4200fe6c: 00ffd646 0000f01d 21004136
Core  1 register dump:
PC      : 0x4200fe70  PS      : 0x00060f30  A0      : 0x00000000  A1      : 0x3fcf7460  
A2      : 0x43f12627  A3      : 0x3fca63c4  A4      : 0x43f12627  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8200fdc9  A9      : 0x3fcf7440  
A10     : 0x00000013  A11     : 0x00000011  A12     : 0x000000ff  A13     : 

[2024-05-25 19:08:36.637]# RECV ASCII>
0x003d0900  
A14     : 0x3c169905  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  

Backtrace: 0x4200fe6d:0x3fcf7460

ELF file SHA256: 3f9246bc6e0febbc

moononournation, 2024-05-25:
Arduino_GFX not support platform.io

mspudi, 2024-05-25:
ttps://registry.platformio.org/libraries/moononournation/GFX%20Library%20for%20Arduino/compatibility

moononournation, 2024-05-25:
Arduino_GFX only support Arduino IDE. If found any issue the occur in platform.io but ok in Arduino IDE, please direct report to platform.io.

closed 2024-06-02 by moononournation