moononournation/Arduino_GFX #485: Error when compiling "PDQgraphicstest" for ESP32_8048S043

eagl1, 2024-05-31:
Hi,

I have ESP32_8048S043 board and when I compile the example `PDQgraphicstest` I get this error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

Invalid library found in E:\Programs_Files\Arduino\libraries\esp32_capacitive_display: no headers files (.h) found in E:\Programs_Files\Arduino\libraries\esp32_capacitive_display

Invalid library found in E:\Programs_Files\Arduino\libraries\lv_port_esp32-master: no headers files (.h) found in E:\Programs_Files\Arduino\libraries\lv_port_esp32-master

Invalid library found in E:\Programs_Files\Arduino\libraries\Nscreen_32-master: no headers files (.h) found in E:\Programs_Files\Arduino\libraries\Nscreen_32-master

Invalid library found in E:\Programs_Files\Arduino\libraries\openHASP-master: no headers files (.h) found in E:\Programs_Files\Arduino\libraries\openHASP-master

In file included from C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino:13:

Arduino_GFX_dev_device.h:177:1: error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32SPIDMA'?

 Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(

 ^~~~~~~~~~~~~~~~~~~~~

 Arduino_ESP32SPIDMA

Arduino_GFX_dev_device.h:185:1: error: 'Arduino_RGB_Display' does not name a type

 Arduino_RGB_Display *gfx = new Arduino_RGB_Display(

 ^~~~~~~~~~~~~~~~~~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'void setup()':

PDQgraphicstest:52:8: error: 'gfx' was not declared in this scope

   if (!gfx->begin())

        ^~~

PDQgraphicstest:58:7: error: 'gfx' was not declared in this scope

   w = gfx->width();

       ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'void loop()':

PDQgraphicstest:150:5: error: 'gfx' was not declared in this scope

     gfx->drawFastHLine(0, i, w, c);

     ^~~

PDQgraphicstest:158:3: error: 'gfx' was not declared in this scope

   gfx->setCursor(0, 0);

   ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'void serialOut(const char*, int32_t, uint32_t, bool)':

PDQgraphicstest:228:5: error: 'gfx' was not declared in this scope

     gfx->fillScreen(BLACK);

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'void printnice(const char*, long int)':

PDQgraphicstest:238:3: error: 'gfx' was not declared in this scope

   gfx->setTextSize(tsb);

   ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testFillScreen()':

PDQgraphicstest:274:3: error: 'gfx' was not declared in this scope

   gfx->fillScreen(WHITE);

   ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testText()':

PDQgraphicstest:286:3: error: 'gfx' was not declared in this scope

   gfx->setCursor(0, 0);

   ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testPixels()':

PDQgraphicstest:379:7: error: 'gfx' was not declared in this scope

       gfx->drawPixel(x, y, gfx->color565(x << 3, y << 3, x * y));

       ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testLines()':

PDQgraphicstest:400:5: error: 'gfx' was not declared in this scope

     gfx->drawLine(x1, y1, x2, y2, BLUE);

     ^~~

PDQgraphicstest:409:5: error: 'gfx' was not declared in this scope

     gfx->drawLine(x1, y1, x2, y2, BLUE);

     ^~~

PDQgraphicstest:420:5: error: 'gfx' was not declared in this scope

     gfx->drawLine(x1, y1, x2, y2, BLUE);

     ^~~

PDQgraphicstest:429:5: error: 'gfx' was not declared in this scope

     gfx->drawLine(x1, y1, x2, y2, BLUE);

     ^~~

PDQgraphicstest:440:5: error: 'gfx' was not declared in this scope

     gfx->drawLine(x1, y1, x2, y2, BLUE);

     ^~~

PDQgraphicstest:449:5: error: 'gfx' was not declared in this scope

     gfx->drawLine(x1, y1, x2, y2, BLUE);

     ^~~

PDQgraphicstest:460:5: error: 'gfx' was not declared in this scope

     gfx->drawLine(x1, y1, x2, y2, BLUE);

     ^~~

PDQgraphicstest:469:5: error: 'gfx' was not declared in this scope

     gfx->drawLine(x1, y1, x2, y2, BLUE);

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testFastLines()':

PDQgraphicstest:487:5: error: 'gfx' was not declared in this scope

     gfx->drawFastHLine(0, y, w, RED);

     ^~~

PDQgraphicstest:491:5: error: 'gfx' was not declared in this scope

     gfx->drawFastVLine(x, 0, h, BLUE);

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testFilledRects()':

PDQgraphicstest:508:5: error: 'gfx' was not declared in this scope

     gfx->fillRect(cx - i2, cy - i2, i, i, gfx->color565(i, i, 0));

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testRects()':

PDQgraphicstest:523:5: error: 'gfx' was not declared in this scope

     gfx->drawRect(cx - i2, cy - i2, i, i, GREEN);

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testFilledCircles(uint8_t)':

PDQgraphicstest:540:7: error: 'gfx' was not declared in this scope

       gfx->fillCircle(x, y, radius, MAGENTA);

       ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testCircles(uint8_t)':

PDQgraphicstest:562:7: error: 'gfx' was not declared in this scope

       gfx->drawCircle(x, y, radius, WHITE);

       ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testFillArcs()':

PDQgraphicstest:576:5: error: 'gfx' was not declared in this scope

     gfx->fillArc(cx1, cy1, i, i - 3, 0, i * r, RED);

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testArcs()':

PDQgraphicstest:589:5: error: 'gfx' was not declared in this scope

     gfx->drawArc(cx1, cy1, i, i - 3, 0, i * r, WHITE);

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testFilledTriangles()':

PDQgraphicstest:604:5: error: 'gfx' was not declared in this scope

     gfx->fillTriangle(cx1, cy1 - i, cx1 - i, cy1 + i, cx1 + i, cy1 + i,

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testTriangles()':

PDQgraphicstest:620:5: error: 'gfx' was not declared in this scope

     gfx->drawTriangle(

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testFilledRoundRects()':

PDQgraphicstest:640:5: error: 'gfx' was not declared in this scope

     gfx->fillRoundRect(cx - i2, cy - i2, i, i, i / 8, gfx->color565(0, i, 0));

     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_50201\PDQgraphicstest.ino: In function 'int32_t testRoundRects()':

PDQgraphicstest:656:5: error: 'gfx' was not declared in this scope

     gfx->drawRoundRect(cx - i2, cy - i2, i, i, i / 8, gfx->color565(i, 0, 0));

     ^~~

exit status 1

'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32SPIDMA'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

moononournation, 2024-06-01:
please use arduino-esp32 2.0.17

eagl1, 2024-06-02:
Now there's 3.0.0, should I install it ?

eagl1, 2024-06-02:
I installed the latest one, same error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, Disabled, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled, Disabled"

In file included from C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_311903\PDQgraphicstest.ino:13:

Arduino_GFX_dev_device.h:177:1: error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?

  177 | Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(

      | ^~~~~~~~~~~~~~~~~~~~~

      | Arduino_ESP32S2PAR8Q

Arduino_GFX_dev_device.h:185:1: error: 'Arduino_RGB_Display' does not name a type

  185 | Arduino_RGB_Display *gfx = new Arduino_RGB_Display(

      | ^~~~~~~~~~~~~~~~~~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_311903\PDQgraphicstest.ino: In function 'void setup()':

PDQgraphicstest:52:8: error: 'gfx' was not declared in this scope

   52 |   if (!gfx->begin())

      |        ^~~

PDQgraphicstest:58:7: error: 'gfx' was not declared in this scope

   58 |   w = gfx->width();

      |       ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_311903\PDQgraphicstest.ino: In function 'void loop()':

PDQgraphicstest:150:5: error: 'gfx' was not declared in this scope

  150 |     gfx->drawFastHLine(0, i, w, c);

      |     ^~~

PDQgraphicstest:158:3: error: 'gfx' was not declared in this scope

  158 |   gfx->setCursor(0, 0);

      |   ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_311903\PDQgraphicstest.ino: In function 'void serialOut(const char*, int32_t, uint32_t, bool)':

PDQgraphicstest:228:5: error: 'gfx' was not declared in this scope

  228 |     gfx->fillScreen(BLACK);

      |     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_311903\PDQgraphicstest.ino: In function 'void printnice(const char*, long int)':

PDQgraphicstest:238:3: error: 'gfx' was not declared in this scope

  238 |   gfx->setTextSize(tsb);

      |   ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_311903\PDQgraphicstest.ino: In function 'int32_t testFillScreen()':

PDQgraphicstest:274:3: error: 'gfx' was not declared in this scope

  274 |   gfx->fillScreen(WHITE);
[... 147 lines cut ...]
C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_311903\PDQgraphicstest.ino: In function 'int32_t testFilledRoundRects()':

PDQgraphicstest:640:5: error: 'gfx' was not declared in this scope

  640 |     gfx->fillRoundRect(cx - i2, cy - i2, i, i, i / 8, gfx->color565(0, i, 0));

      |     ^~~

C:\Users\eagle\AppData\Local\Temp\arduino_modified_sketch_311903\PDQgraphicstest.ino: In function 'int32_t testRoundRects()':

PDQgraphicstest:656:5: error: 'gfx' was not declared in this scope

  656 |     gfx->drawRoundRect(cx - i2, cy - i2, i, i, i / 8, gfx->color565(i, 0, 0));

      |     ^~~

exit status 1

'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

eagl1, 2024-06-08:
Yep, you're right. esp32 2.0.17 solved the problem but nothing shows on the display. It's black.

eagl1, 2024-06-18:
Hi,

I'm still getting this error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, OPI PSRAM, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled, Disabled"

In file included from E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino:11:

Arduino_GFX_dev_device.h:14:1: error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?

   14 | Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(

      | ^~~~~~~~~~~~~~~~~~~~~

      | Arduino_ESP32S2PAR8Q

Arduino_GFX_dev_device.h:22:1: error: 'Arduino_RGB_Display' does not name a type

   22 | Arduino_RGB_Display *gfx = new Arduino_RGB_Display(

      | ^~~~~~~~~~~~~~~~~~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'void setup()':

GFX_Library_for_Arduino:50:8: error: 'gfx' was not declared in this scope

   50 |   if (!gfx->begin())

      |        ^~~

GFX_Library_for_Arduino:56:7: error: 'gfx' was not declared in this scope

   56 |   w = gfx->width();

      |       ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'void loop()':

GFX_Library_for_Arduino:148:5: error: 'gfx' was not declared in this scope

  148 |     gfx->drawFastHLine(0, i, w, c);

      |     ^~~

GFX_Library_for_Arduino:156:3: error: 'gfx' was not declared in this scope

  156 |   gfx->setCursor(0, 0);

      |   ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'void serialOut(const char*, int32_t, uint32_t, bool)':

GFX_Library_for_Arduino:226:5: error: 'gfx' was not declared in this scope

  226 |     gfx->fillScreen(BLACK);

      |     ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'void printnice(const char*, long int)':

GFX_Library_for_Arduino:236:3: error: 'gfx' was not declared in this scope

  236 |   gfx->setTextSize(tsb);

      |   ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'int32_t testFillScreen()':

GFX_Library_for_Arduino:272:3: error: 'gfx' was not declared in this scope
[... 149 lines cut ...]
E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'int32_t testFilledRoundRects()':

GFX_Library_for_Arduino:638:5: error: 'gfx' was not declared in this scope

  638 |     gfx->fillRoundRect(cx - i2, cy - i2, i, i, i / 8, gfx->color565(0, i, 0));

      |     ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'int32_t testRoundRects()':

GFX_Library_for_Arduino:654:5: error: 'gfx' was not declared in this scope

  654 |     gfx->drawRoundRect(cx - i2, cy - i2, i, i, i / 8, gfx->color565(i, 0, 0));

      |     ^~~

exit status 1

'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

moononournation, 2024-06-18:
please use arduino-esp32 v2.x, v3.0 will have this error.

eagl1, 2024-06-18:
Yep, it worked.

[image]

eagl1, 2024-06-18:
I have many questions, but I will ask this one:

Why did you declared these pointers at the end of the `Arduino_GFX_Library.h` ?

```
Arduino_DataBus *create_default_Arduino_DataBus();
Arduino_GFX *create_default_Arduino_GFX();
```

And what are they exactly ?

are they pointers to class constructor ?

if so then why you didn't declared other ones for the other two classes `Arduino_ESP32RGBPanel` and `Arduino_RGB_Display` ?

eagl1, 2024-06-18:
Another question:

Where is the driver library for `ST7262` ? I didn't find it in `\display` folder.