moononournation/Arduino_GFX #486: Problem with jpeg_dec_parse_header()

woaeo, 2024-05-31:
hello, i have an issue, when i run example ImgViewerJpeg_ESP32_JPEG code breake at jpeg_dec_parse_header(), end my esp is constantly rebooting with this error:
[image]
i didnt find answer in jpeg decode library repo and i have no idea what to do with this problem...

i need some help ,.,

moononournation, 2024-05-31:
try another jpeg file

woaeo, 2024-05-31:
omg, it kinda helped, but now it stuck on jpeg_dec_process() and changing file agai didnt helped

moononournation, 2024-05-31:
is it normal if you use original jpeg file in data folder?

woaeo, 2024-06-01:
no, when i use example file from data folder it again stuck at jpeg_dec_parse_header() with the same "Core  1 panic'ed (StoreProhibited)."

moononournation, 2024-06-01:
then use jpegdec instead

closed 2024-06-02 by moononournation

woaeo, 2024-06-02:
thats worked, i guess im gonna use it from now

thanks