moononournation/Arduino_GFX #488: using XL9535 IOs

s54b32julian, 2024-06-02:
I want to using the Port1 of the XL9535 Port expander (for general purpose digital Input or Outputs), that is used in databus class Arduino_XL9535SWSPI. But i´m not sure how to access it from outside.
All the correct functions are already there like PinMode, digitalRead/Write etc. but how can i call them in the main program to access them?

s54b32julian, 2024-06-05:
I tried to add the members pinmode and digital read from xl9535 class to the Arduino databus (just route them through), compiles fine but esp crashes during there usage. 
Any suggestions?

moononournation, 2024-06-05:
I don't think you need a GFX library to indirect operate the IO expander. Why don't you direct call their API?

closed 2024-06-05 by moononournation