moononournation/Arduino_GFX #492: Problems with using setRotation

anla-xu, 2024-06-07:
Hello, when I use Arduino_ST7789 to drive a 128X128 screen, since the screen is fixed and rotated 180° counterclockwise, I output a string on the screen, but the image is inverted. I want the string to be rotated 180° to display normally. After using setRotation(2), I don't get what I want. How can I solve this problem? Thank you

anla-xu, 2024-06-07:
[image]
like this

moononournation, 2024-06-07:
st7789 did not have 128x128 display, wild guess the display is st7735

anla-xu, 2024-06-07:
Is it the wrong screen type that causes the screen flipping abnormality?

anla-xu, 2024-06-07:
It is indeed the abnormality caused by the wrong IC. After changing to the corresponding IC driver library, it can be displayed normally. Thank you.

closed 2024-06-07 by anla-xu