moononournation/Arduino_GFX #498: GFX display not working with T-Display S3 AMOLED

Zitus, 2024-06-16:
Is this a correct basic sketch for LilyGo T-Display S3 Amoled? I tried many times and didn't work. It compiles and uploads correctly. 

#include <Arduino_GFX_Library.h>

#define GFX_DEV_DEVICE LILYGO_T_DISPLAY_S3_AMOLED
Arduino_DataBus *bus = new Arduino_ESP32QSPI(
    6 /* cs */, 47 /* sck */, 18 /* d0 */, 7 /* d1 */, 48 /* d2 */, 5 /* d3 */);
Arduino_GFX *gfx = new Arduino_RM67162(bus, 17 /* RST */, 0 /* rotation */);

void setup(void)
{
  Serial.begin(115200);
  Serial.println("Setup Started");

#ifdef GFX_EXTRA_PRE_INIT
  GFX_EXTRA_PRE_INIT();
#endif

  // Init Display
  if (!gfx->begin())
  {
    Serial.println("gfx->begin() failed!");
    while (1); // Halt the program
  }
  Serial.println("gfx->begin() success!");

  gfx->fillScreen(BLACK);
  Serial.println("Screen filled with BLACK");

#ifdef GFX_BL
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);
  Serial.println("Backlight turned ON");
#endif

  gfx->setCursor(10, 10);
  gfx->setTextColor(RED);
  gfx->println("Hello World!");
  Serial.println("Hello World printed!");

  delay(5000); // 5 seconds
}

void loop()
{
  gfx->setCursor(random(gfx->width()), random(gfx->height()));
  gfx->setTextColor(random(0xffff), random(0xffff));
  gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
  gfx->println("Hello World!");

  delay(1000); // 1 second
}

moononournation, 2024-06-17:
it have many similar products, can you give your purchase link to check?

Zitus, 2024-06-17:
https://www.lilygo.cc/products/t-display-s3-amoled?variant=43506902368437

moononournation, 2024-06-17:
the pins are correct, can you run PDQgraphicstest ok?

Zitus, 2024-06-17:
Correct device was uncommented in Arduino_GFX_dev_device.h , (don't know
about pins) but  PDQgraphicstest produces nothing on the board.

moononournation, 2024-06-17:
any output in serial monitor?

Zitus, 2024-06-17:
No, nothing in serial.

How do I know what to uncomment in databus or display, if there's no name
about the board?

moononournation, 2024-06-17:
it should at least have some output show program started, please check youur board parameters selected correct in Arduino IDE tools menu

Zitus, 2024-06-17:
> Sorry, there is output.
>
> Pixels	3935183
>
> Benchmark	micro-secs
>
> Screen fill	38491
>
> Text	55165
>
> Pixels	3935201
>
> Lines	3528155
>
> Horiz/Vert Lines	8489
>
> Rectangles (filled)	49674
>
> Rectangles (outline)	7060
>
> Triangles (filled)	120497
>
> Triangles (outline)	115618
>
> Circles (filled)	201453
>
> Circles (outline)	348419
>
> Arcs (filled)	111537
>
> Arcs (outline)	229778
>
> Rounded rects (filled)	90156
>
> Rounded rects (outline)	76782
>
> Done!
>
>
> But the screen remains black

Can you specify what to uncomment
in Arduino_GFX_databus.h, Arduino_GFX_display.h, and Arduino_GFX_pins.h for
this board?

moononournation, 2024-06-17:
the output seems correct. if you uncommented the dev device in the header file, no need to touch any parameters

Zitus, 2024-06-17:
I'm using arduino ide 2.1.1
Board is new, when connected first time screen worked and showed graphics.
line uncommented in dev device is #define LILYGO_T_Display_S3_AMOLED

Any other ideas?

Zitus, 2024-06-17:
Display working now. Any information on how to use sprites with Arduino GFX Library?

moononournation, 2024-06-17:
good new, can you share what fix it?
and you may refer SpriteGif.ino to get some hint.

closed 2024-06-17 by moononournation

Zitus, 2024-06-17:
Thanks.

Yes , you need to add the library LilyGo-AMOLED-Series-master.zip and not
other.

moononournation, 2024-06-17:
Oh really?