moononournation/Arduino_GFX #502: Adding a new 7" blue EA7013 7inch ESP32 display with 16-bit parallel RGB interface

eagl1, 2024-06-29:
Hi,

I just received this 7" blue EA7013 7inch ESP32 display with 16-bit parallel RGB interface.
It came with the LVGL demo.

Aliepxress link:
[https://vi.aliexpress.com/item/1005006484114641.html?src=google&src=google&albch=shopping&acnt=897-548-8760&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&src=google&albch=shopping&acnt=897-548-8760&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=ar1005006484114641&ds_e_product_merchant_id=5317711677&ds_e_product_country=SA&ds_e_product_language=ar&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=20214492666&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&gclid=CjwKCAjw-O6zBhASEiwAOHeGxYTqRCD0ri50AKZJN47j9Jb9suMipwUmJOD2MX0R7VtIaGmwPmQHkRoCAE0QAvD_BwE&aff_fcid=aa37169580324b0d9f14e76adc91d4a6-1719451083445-06895-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=aa37169580324b0d9f14e76adc91d4a6-1719451083445-06895-UneMJZVf&terminal_id=eccb76f812b4466980f94cb30194a34d&afSmartRedirect=y#nav-specification](url)

This is the display schematic file:
[Uploading EA7013.pdf…]()

I tried to modify the pin arrangement in the object definition to this in this `Arduino_GFX_dev_device.h` file in the demo code:
```
// #define ESP32_8048S043 // this one worked with esp32 2.0.17 board package
// #define ESP32_8048S070
 #define ESP32_EA7013

#if defined(ESP32_8048S043)
#define GFX_DEV_DEVICE ESP32_8048S043
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
    0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

#elif defined(ESP32_8048S070)
#define GFX_DEV_DEVICE ESP32_8048S070
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

#elif defined(ESP32_EA7013)
#define GFX_DEV_DEVICE ESP32_EA7013
#define GFX_BL 1
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    5 /* DE */, 2 /* VSYNC */, 4 /* HSYNC */, 6 /* PCLK */,
    7 /* R3 */, 15 /* R4 */, 16 /* R5 */, 8 /* R6 */, 3 /* R7 */,
    46 /* G2 */, 9 /* G3 */, 10 /* G4 */, 11 /* G5 */, 12 /* G6 */, 13 /* G7 */,
    14 /* B3 */, 21 /* B4 */, 47 /* B5 */, 48 /* B6 */, 45 /* B7 */,
    0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);
```

I set the pin mapping according to the schematic diagram.
It didn't work, only the backlight worked:

What I'm missing here ?

moononournation, 2024-07-02:
I did not have this display in hand. In general, try adjust the parameters according to the display data sheet.

closed 2024-07-02 by moononournation

eagl1, 2024-07-02:
I'm trying but it's not working until this moment. Only the backlight worked.

I know it will need more time to get it to work eventually.

eagl1, 2024-07-03:
I got it to work in ESP-IDF platform, easy RGB example code, only changed the RGB parameters and worked perfectly.

But I don't know what is the difference between your RGB code and the one in the ESP-IDF plattform ? it's almost the same code.

eagl1, 2024-07-06:
I have a question, if I wanted to copy `Arduino_ESP32RGBPanel` library and use it in `TFT_eSPI` library. What requirements I have to use along with it ?