moononournation/Arduino_GFX #506: Arduino_Canvas Compilation error: no matching function for call to 'Arduino_Canvas::Arduino_Canvas(int, int, Arduino_ESP32RGBPanel*&, int, int, int)'

vitasynergy, 2024-07-14:
Hello,
I've a working app on a Makerfabs 4.3" TFT Parallel with touch using your "Arduino_GFX_Library.h" and Bodmer's Jpegdecoder (https://github.com/bitbank2/JPEGDEC.git).

The issue here is that there's no way to use the Arduino_Canvas (prototype here https://github.com/moononournation/Arduino_GFX/blob/master/src/canvas/Arduino_Canvas.h) as to handle a canvas instead of displaying each single jpg image one after the other.

Here's the declaration sequence that gets the "NO MATCHING FUNCTION FOR CALL TO ARDUINO_CANVAS":
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
  45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
  5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
  8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
  0 /* hsync_polarity */, 40 /* hsync_front_porch */, 48 /* hsync_pulse_width */, 88 /* hsync_back_porch */,
  0 /* vsync_polarity */, 13 /* vsync_front_porch */, 3 /* vsync_pulse_width */, 32 /* vsync_back_porch */,
  1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
  800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);
Arduino_GFX *gfxCanvas = new Arduino_Canvas(800, 480, rgbpanel, 0, 0, 0);

Btw, throwing the Arduino_Canvas constructor the code gets compiled correctly and the app works.

Thanks

P.S.the ST7282 lcd driver that seems to be integrated into the 4.3" Makerfabs does not appears in the list of supported displays although the Makerfabs 4.3" Parallel TFT with touch appears in the supported dev devices.....

moononournation, 2024-07-17:
Please use arduino-esp32 2.0.17

closed 2024-07-17 by moononournation

vitasynergy, 2024-07-17:
switched to arduino-esp32 2.0.17 and the Canvas declaration continue to throw "no matching function for call to 'Arduino_Canvas::Arduino_Canvas(int, int, Arduino_ESP32RGBPanel*&, int, int, int)'.

The only way to get compiled the Canvas declaration is to use : 
Arduino_Canvas *gfxCanvas = new Arduino_Canvas(800, 480, gfx);

Though I need to test it yet. I'll test it and update this post