moononournation/Arduino_GFX #507: Color problem while using ST7701 4" screen

prof955, 2024-07-15:
ESP32-S3 + 4" ST7701 screen.
PDQgraphicstest works without problem.

When i use library to draw something like circle etc. colors are ok.
But when i try to draw Jpeg/Png/gif examples, drawing is ok but colors are not.
Not inverted colors. i tried invert, it didn't solved. See photo of screen and original image.
Problem is same for every img example like PNG, jpeg, gif bmp.

[image]
[image]

[image]
[image]

moononournation, 2024-07-17:
try switching between draw16bitBeRGBBitmap() and draw16bitRGBBitmap().

closed 2024-07-17 by moononournation

prof955, 2024-07-17:
Unfurtunatelly didn't solved problem. Colors changed tho. but changed to another weird colors.
See below:

draw16bitBeRGBBitmap:
[image]
[image]

draw16bitRGBBitmap:
[image]
[image]

moononournation, 2024-07-17:
seem the rgb pins mapping not correct

prof955, 2024-07-17:
They are exactly like the PDQexample.

#define GFX_BL 38// Back Light pini

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    39 /* CS */, 48 /* SCK */, 47 /* SDA */,
    18 /* DE */, 17 /* VSYNC */, 16 /* HSYNC */, 21 /* PCLK */,
    4 /* R0 */, 3 /* R1 */, 2 /* R2 */, 1 /* R3 */, 0 /* R4 */,
    10 /* G0 */, 9 /* G1 */, 8 /* G2 */, 7 /* G3 */, 6 /* G4 */, 5 /* G5 */,
    15 /* B0 */, 14 /* B1 */, 13 /* B2 */, 12 /* B3 */, 11 /* B4 */
);
// Uncomment for 4" rect display
Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel(
    bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */,
    true /* IPS */, 480 /* width */, 480 /* height */,
    st7701_type1_init_operations, sizeof(st7701_type1_init_operations),
     true /* BGR */,
     10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
     10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */);
     
     
     This is what manufacturer info:
     
[image]

moononournation, 2024-07-17:
st7701_type1_init_operations may not match, had you tried other type number?

prof955, 2024-07-17:
So RGB pins are ok then?

i didn't try that. How can I change it?

How to? Just change init like this?: st7701_type2_init_operations

şunu yazdı:

prof955, 2024-07-18:
i tried type 2 init. Screen became darker and screen is flickering BUT colors are same. So for the issue it didn't solved. 
Check screenshot pls:

[image]

IDK how github works but why top left is set to "closed" while issue is not solved or something?
i really don't know :) is there no on-going, under investigation ro something like that?

Postlast, 2024-09-25:
I guess you are using GFX_Library_for_Arduino and not the Arduino_GFX-master library supplied with the 4.0inch_ESP32-4848S040.zip example. Or replace the library. The files Arduino_ST7701_RGBPanel.cpp and Arduino_ST7701_RGBPanel.h are different. You can change it in Arduino_ST7701_RGBPanel.h
`static const uint8_t st7701_type1_init_operations[] = {
BEGIN_WRITE,
WRITE_COMMAND_8, 0xFF,
WRITE_BYTES, 5, 0x77, 0x01, 0x00, 0x00, 0x10,

WRITE_C8_D16, 0xC0, 0x3B, 0x00,
WRITE_C8_D16, 0xC1, 0x0D, 0x02,
WRITE_C8_D16, 0xC2, 0x31, 0x05, 
WRITE_C8_D8, 0xCD, _**0x00,//0x08**_`
 set Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel( ........ true /*false IPS */,
 and #define LV_COLOR_16_SWAP 1 in lv_conf.h