moononournation/Arduino_GFX #516: ESP32-8048S070 - 7 Inch Display not working

Ameso-Karthi, 2024-08-12:
Right now esp32 board updated on v 3.0.3 so its showing lot's of error if i running on the example function and this is my code

/*******************************************************************************
 ******************************************************************************/
#include <Arduino_GFX_Library.h>

#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin
#define TFT_BL 2
/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
//Arduino_DataBus *bus = create_default_Arduino_DataBus();

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
//Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

#endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{
    gfx->begin();
    gfx->fillScreen(BLACK);

#ifdef TFT_BL
    pinMode(TFT_BL, OUTPUT);
    digitalWrite(TFT_BL, HIGH);
#endif

    gfx->setCursor(10, 10);
    gfx->setTextColor(RED);
    gfx->println("Hello World!");

    delay(2000); // 5 seconds
}

void loop()
{
    gfx->setCursor(random(gfx->width()), random(gfx->height()));
    gfx->setTextColor(random(0xffff), random(0xffff));
    gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
    gfx->println("Hello World!");

    delay(1000); // 1 second
}

this is the error it's showing

C:\Users\karth\OneDrive\Documents\Arduino\ESP32_S3_LVGL_7_INCH_DEMO_TEST\ESP32_S3_LVGL_7_INCH_DEMO_TEST.ino:18:1: error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?
   18 | Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
      | ^~~~~~~~~~~~~~~~~~~~~
      | Arduino_ESP32S2PAR8Q
C:\Users\karth\OneDrive\Documents\Arduino\ESP32_S3_LVGL_7_INCH_DEMO_TEST\ESP32_S3_LVGL_7_INCH_DEMO_TEST.ino:25:1: error: 'Arduino_RGB_Display' does not name a type
   25 | Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
      | ^~~~~~~~~~~~~~~~~~~
C:\Users\karth\OneDrive\Documents\Arduino\ESP32_S3_LVGL_7_INCH_DEMO_TEST\ESP32_S3_LVGL_7_INCH_DEMO_TEST.ino: In function 'void setup()':
C:\Users\karth\OneDrive\Documents\Arduino\ESP32_S3_LVGL_7_INCH_DEMO_TEST\ESP32_S3_LVGL_7_INCH_DEMO_TEST.ino:35:5: error: 'gfx' was not declared in this scope
   35 |     gfx->begin();
      |     ^~~
C:\Users\karth\OneDrive\Documents\Arduino\ESP32_S3_LVGL_7_INCH_DEMO_TEST\ESP32_S3_LVGL_7_INCH_DEMO_TEST.ino: In function 'void loop()':
C:\Users\karth\OneDrive\Documents\Arduino\ESP32_S3_LVGL_7_INCH_DEMO_TEST\ESP32_S3_LVGL_7_INCH_DEMO_TEST.ino:52:5: error: 'gfx' was not declared in this scope
   52 |     gfx->setCursor(random(gfx->width()), random(gfx->height()));
      |     ^~~

exit status 1

Compilation error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?

moononournation, 2024-08-12:
Please use arduino-esp32 v2.x

closed 2024-08-21 by moononournation

ZinggJM, 2024-10-20:
I have added an updated version of Arduino_RGB_Display and Arduino_ESP32RGBPanel with support for ESP32 3.0.5 to my fork.
https://github.com/ZinggJM/Arduino_GFX
I don't create a pull request yet, as this version is experimental (I am no expert for this ESP32S3 feature).
-jz-

moononournation, 2024-10-20:
great, waiting for your contribution.

ZinggJM, 2024-10-20:
Thank you! 
Do you intend to take a look at it first, or should I submit a pull request?
Alternately, you can copy the sources and adapt it to your preferences.
Or someone with experience with this feature could take a look at it.