moononournation/Arduino_GFX #519: Support Request: Implement HX8369A 4.3-inch TFT Display Driver for ESP32-WROOM-32 in Arduino GFX library

EMPERO-CYBER-HUB, 2024-08-17:
I am attempting to interface an **HX8369A** 4.3-inch TFT display with an ESP32-WROOM-32 module using the [image]version 1.4.7. Despite following the setup instructions and connecting the necessary pins, the display does not show any output. Below is my setup and code, along with details of the interface.

Hardware Details:
- Display: HX8369A 4.3-inch TFT (480x800 resolution)
- Microcontroller: ESP32-WROOM-32
- Library: Arduino GFX Library v1.4.7
- Power: The RD pin is connected to 3.3V as suggested, to pull it up since it's not used.

Pin Configuration:
- TFT_DC (Data/Command): GPIO 27
- TFT_CS (Chip Select): GPIO 5
- TFT_WR (Write): GPIO 26
- TFT_RD (Read - Pulled up to 3.3V, not used): GFX_NOT_DEFINED (or 3.3V directly)
- TFT_RST (Reset): GPIO 33
- Data Bus:
  - D0: GPIO 16
  - D1: GPIO 17
  - D2: GPIO 18
  - D3: GPIO 19
  - D4: GPIO 21
  - D5: GPIO 12
  - D6: GPIO 23
  - D7: GPIO 25

Code Used:
`
#include <Arduino_GFX_Library.h>

#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin
#define TFT_RST 33

#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */
Arduino_DataBus *bus = new Arduino_ESP32PAR8(27 /* DC */, 5 /* CS */, 26 /* WR */, GFX_NOT_DEFINED /* RD */,
    16 /* D0 */, 17 /* D1 */, 18 /* D2 */, 19 /* D3 */, 21 /* D4 */, 12 /* D5 */, 23 /* D6 */, 25 /* D7 */);

Arduino_GFX *gfx = new Arduino_HX8369A(bus, TFT_RST, 0 /* rotation */, false /* IPS */, 480, 800, 0, 7, 0, 57);
#endif /* !defined(DISPLAY_DEV_KIT) */

void setup(void)
{
  Serial.begin(115200);
  Serial.println("Arduino_GFX Hello World example");

  if (!gfx->begin())
  {
    Serial.println("gfx->begin() failed!");
  }
  gfx->fillScreen(BLACK);

  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);

  gfx->setCursor(10, 10);
  gfx->setTextColor(RED);
  gfx->println("Hello World!");

  delay(5000); // 5 seconds
}

void loop()
{
  gfx->setCursor(random(gfx->width()), random(gfx->height()));
  gfx->setTextColor(random(0xffff), random(0xffff));
  gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
  gfx->println("Hello World!");

  delay(1000); // 1 second
}
`

Problem:
The display remains unresponsive and shows no output. The initialization seems to proceed, but nothing appears on the screen.

Potential Areas of Concern:
1. Pin Configuration: The pin mapping used may not be correct or fully supported.
2. Display Driver Support: The HX8369A driver may not be fully supported or configured correctly in the library.
3. Pull-up on RD Pin: The RD pin is pulled up to 3.3V as recommended since it's not used. This should be correct, but there might be issues related to this.

Request:
Could you please provide guidance on whether the pin configuration or the driver support needs adjustment in the Arduino GFX library? Additionally, if there are any examples or debugging steps you can recommend for getting the HX8369A display working with the ESP32, that would be highly appreciated.

---

moononournation, 2024-08-17:
had you pull up RD pin?

EMPERO-CYBER-HUB, 2024-08-17:
Nice to have you my man, i connect the RD pin with the  3.3v from my esp32

moononournation, 2024-08-17:
I only know this common mistake I always do, no further idea.

EMPERO-CYBER-HUB, 2024-08-17:
ok but please Help me , Am i right on the Wiering and the code if so please confirm, i have a timely project i have to submit and unless i solve this issue i can't continue to the other. (I watched your 2 YouTube Videos Which Works great so please tell me where i made mistake)

moononournation, 2024-08-17:
um... if your ESP32 have PSRAM, please avoid using pins 16 and 17

closed 2024-08-17 by moononournation

EMPERO-CYBER-HUB, 2024-08-17:
I have  disabled the PSRAM , and also changed the pins to this 

```/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_DataBus *bus = new Arduino_ESP32PAR8(27 /* DC */, 5 /* CS */, 26 /* WR */, GFX_NOT_DEFINED /* RD */,
    34 /* D0 */, 35 /* D1 */, 18 /* D2 */, 19 /* D3 */, 21 /* D4 */, 12 /* D5 */, 23 /* D6 */, 25 /* D7 */);
    ```
    But still the display not working

EMPERO-CYBER-HUB, 2024-08-17:
@moononournation you are the owner of this library please you can help for real , i have bought many of this type of TFT so it will be wast if i can't make them work ,

EMPERO-CYBER-HUB, 2024-08-17:
hi anyone here to help